OpenBCI/OpenBCI_GUI #272: Not getting Trigger data using wifi shield connected to Cyton and daisy combination

msmadan, 2017-10-24:
I added a Digital trigger on D12 using the procedure from this link

http://docs.openbci.com/OpenBCI Software/04-OpenBCI_Cyton_SDK#openbci-cyton-sdk-firmware-v300-new-commands-board-mode

I am not getting any trigger data stored when I checked the saved data file.
I tried both D12 and D11 pins.

Any help you can provide in this regard is much appreciated
-Raja

msmadan, 2017-10-27:
@aj-ptw , 
Can you please let me know when you will  get a chance to take a quick look at this issue
Thanks for all the help
Raja

aj-ptw, 2017-10-27:
@msmadan 

Hi, I'm sorry for the delay, was busy releasing the new [v3.1.0](https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.1.0).

Let's solve this! Okay so how are you sending the command? Do you still see accel data?

How about you upload 50 or so samples to a comment so i can see how the files are ending.

Thanks so much for your time!

aj-ptw, 2017-10-27:
sorry didn't mean to close

aj-ptw, 2017-10-27:
what code did you do to add this?

msmadan, 2017-10-27:
I used  /3 for digital ,
I was not getting any accelerometer data 
I have attached the data I got with this

[OpenBCI-RAW-2017-10-22_17-14-40.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/1423741/OpenBCI-RAW-2017-10-22_17-14-40.txt)

aj-ptw, 2017-10-28:
@msmadan cool sounds like some data is not making it's way through the data writing.

You are my favorite user! Sorry for the bugs!

msmadan, 2017-10-28:
I will change my user handle to "bug catcher",  ;)
Happy to help
and thanks for your help
Raja

aj-ptw, 2017-10-28:
which gui version are you using?

aj-ptw, 2017-10-28:
is this over wifi or bluetooth?

msmadan, 2017-10-28:
GUI version 3.1.0
I was trying this over wifi

msmadan, 2017-10-28:
GUI version 3.1.0 beta1
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.1.0-beta1

msmadan, 2017-10-28:
Sorry 
Version 3.1.0-beta 2
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.1.0-beta2

aj-ptw, 2017-10-28:
hey this is totally just because this feature has not been coded yet! So im not calling it a bug, i just wrote a bunch of code for this. and you can see the new code above. Working on this new code!

aj-ptw, 2017-10-28:
You should be using 3.1.0 now by the way, lots of bugs fixes!

aj-ptw, 2017-10-28:
I’m making progress, almost there!

msmadan, 2017-10-30:
Hi AJ 
were you able to finish the code 
Raja

aj-ptw, 2017-10-30:
Well, it's in my [development branch](https://github.com/aj-ptw/OpenBCI_GUI/tree/development), but it's not complete yet. You can see the code reference up above, I will continue to tag this issue when i make commits, but I ran out of time, this is a bigger feature, not an overnight patch.

aj-ptw, 2017-10-30:
Please help and contribute to via a pull request to my development branch if you test the code and it doesn't work. I made two new widgets for you! One for analog and one for digital.

msmadan, 2017-10-31:
Hi AJ
Sorry to be a pain
I am new to Processing
I tried to run your GUI development program , I get this error ("String(int) does not exist)
I have attached an image of this. Can you please let me know where to fix this in the code

[image]

-Raja

aj-ptw, 2017-10-31:
That's so cool you're new to processing! It's a decent language wrapper over Java, very cool.

Let's see the easiest way to fix this is remove `String` function call from each of these output.print lines. Look at line 292.

If that doesn't work then switch to have two output.print() calls per line, i.e. the digital would look like
```java
output.print(", "); output.print(data.auxValues[0] & 0xFF);
output.print(", "); output.print((data.auxValues[0] & 0xFF00) >> 8);
output.print(", "); output.print((data.auxValues[1] & 0xFF00) >> 8);
```

msmadan, 2017-10-31:
that fixed it, Thanks
Was able to run the GUI, I  will check if I am able to get aux data and let you  know 
[image]

aj-ptw, 2017-10-31:
Cool!! Submit a pull request to my development branch please! [Learn to submit a pull request.](https://help.github.com/articles/creating-a-pull-request/)

msmadan, 2017-10-31:
Is there a tutorial you can point me to create a binary for windows 10 or can please let me know how to do it
-Raja

msmadan, 2017-10-31:
Never done a pull, will learn and do it soon
-Raja

aj-ptw, 2017-10-31:
I opened #276 to add a guide for building binaries to the next release.

My best advice for learning git:
**If you hit a problem with git, spend your time finding someone who knows git to explain and solve the problem rather then try to solve the problem yourself. It's tough at first and takes time not intelligence to learn.**

msmadan, 2017-10-31:
I used Export application command to export as binary and I am using the Hub from Version 3.1.0.(is this correct ?)

aj-ptw, 2017-10-31:
that's what i do too!

msmadan, 2017-10-31:
OK ,Thanks, I will test it with the board and let know the results

msmadan, 2017-10-31:
Hi AJ
I tried the new program ,The digital widget was able to change the board to digital mode
But the digital aux data is not saving.
I have attached data.Is there any line in code you like me check.
[OpenBCI-RAW-2017-10-31_19-34-45.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/1432629/OpenBCI-RAW-2017-10-31_19-34-45.txt)
-Raja

aj-ptw, 2017-11-01:
@msmadan okay development seems to be printing to output file now! give it a shot, pull my development branch.

msmadan, 2017-11-01:
Thanks AJ , I will test it tonite 
Raja

aj-ptw, 2017-11-01:
ok i am working on improving it.

msmadan, 2017-11-01:
I just pulled yur dev branch, if do any improvements please let me know i will pull again

aj-ptw, 2017-11-02:
Hi I’m nervous I only fixed on 8 channel for some reason. Will test with 16 in a bit here

msmadan, 2017-11-02:
ok thanks

aj-ptw, 2017-11-02:
Ok this is in the hub. Tracking this down now. I was testing my new digital read widgets and was getting so mad they weren't working and then realized i was duplicating your bug! The data is just not being sent from the hub! Easy fix :)

aj-ptw, 2017-11-02:
i just pushed fixes for daisy data and am building a new hub to send daisy aux data to the gui, it was never being sent, this could be the reason you have not been able to get it! you should also pull my [aj-ptw/development](https://github.com/aj-ptw/OpenBCI_GUI/tree/development) branch

@msmadan and @gerrievanzy l think y'all use windows right? 

Well as soon as these windows builds are done, you can select your operating system x64 or x86 and then hit the artifact tab and download the zip file and there is your new hub to solve this problem. https://ci.appveyor.com/project/aj-ptw/openbci-ganglion-electron/build/1199

msmadan, 2017-11-02:
@aj-ptw , yes I am using windows 10.
So if I am reading this correctly I need to pull the new  GUI from your development branch 
and also update the Hub from your link? please confirm

gerrievanzyl, 2017-11-02:
@aj-ptw I use OSX

gerrievanzyl, 2017-11-02:
@aj-ptw Did you have a chance to check out the other issues I have been working on?  Not sure how to send a pull request to your personal repo.  I have also included  a new MarkerMode widget

aj-ptw, 2017-11-02:
> So if I am reading this correctly I need to pull the new GUI from your development branch
and also update the Hub from your link? please confirm

Yes.

@gerrievanzyl here i made this release for you [v1.3.5-beta1](https://github.com/OpenBCI/OpenBCI_Ganglion_Electron/releases/tag/v1.3.5-beta1)

aj-ptw, 2017-11-02:
> @aj-ptw Did you have a chance to check out the other issues I have been working on? Not sure how to send a pull request to your personal repo. I have also included a new MarkerMode widget

I made a gif for you!
[image]

Also some inspiration!
[image]

gerrievanzyl, 2017-11-02:
@aj-ptw

I did some extensive testing.

Bottom line - the system is very usable but there are some small issues:

2. The markers (I expect the aux channels as well)  are coming back and being saved in the file (Hooray!)
1. the system is much, much more stable (only once in >30 tests did the data stream freeze) but there are still a few small issues
2. BLE is not a suitable channel for receiving data with markers - sending the two character marker command causes the next packet (the one with the marker) to be lost 8/10 times.  I do not think we need to work on this as the file on SD card contains all the information.
3. The SD Card is not entirely stable.  For example a repeatable bug is when you try to save to the SD Card without the SD Card present.  When you hotplug the SD Card it corrupts the card and you need to reformat the SD Card and restart both the GUI/Hub and the board

Below is a log of the testing. 

# OpenBCI Test
Electron Hub v1.3.5-beta1
OpenBCI_GUI gerrievanzy/OpenBCI_GUI feature/UDPmarker

## BLE Test

1. 16 Channels BLE - 10 markers sent 3 received
2. 16 Channels BLE - 10 markers sent 5 received
3. 16 Channels BLE - 10 markers sent 3 received
4. 16 Channels BLE - 10 markers sent 0 received (178 lines in file???)
5. 16 Channels BLE - 10 markers sent 5 received

**Conclusion:  There is something seriously broken with the BLE marker sending/receiving.  When a maker signal is sent to the board, 8/10 times it results in a dropped packet.  It seems that often it is the return packet with the marker that is dropped.  However, the file on the SD card reliably contains all the makers.**

## Wifi Test
(only send markers when the wifi light is solid - otherwise markers may be lost in packets)

1. 16 Channels Wifi - 10 markers sent 10 received
2. 16 Channels Wifi - 10 markers sent 9 received (board froze at end)
3. 16 Channels Wifi - 10 markers sent 10 received no SD Card Write
4. 16 Channels Wifi - 10 markers sent 10 received SD Card Write
5. 16 Channels Wifi - 10 markers sent 9 received SD Card Write (wifi connection issues)
6. 16 Channels Wifi - 10 markers sent 10 received SD Card Write
7. 16 Channels Wifi - 10 markers sent 10 received no SD Card Write (fresh run)
8. 16 Channels Wifi - 10 markers sent 10 received no SD Card Write (2nd run)
9. 16 Channels Wifi - 10 markers sent 10 received activate SD Card Write -but no SD card present (3rd run)
10. 16 Channels Wifi - 10 markers sent 10 received activate SD Card Write -but no SD card present (4th run)

**Inserted SD Card**

Hub: processConnect: made it -- 530430
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
OpenBCI_GUI: stopRunning: stop running...
Closing any open SD file. Writing 'j' to OpenBCI.
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
OpenBCI_GUI: stopRunning: stop running...
Closing any open SD file. Writing 'j' to OpenBCI.
Failed to connect to data source...

**Tested SD Card - all good
Inserted SD Card**

Powering 3 widgets up, and remapping.
Activating widget [0]
Activating widget [1]
Activating widget [2]
  3c -- 701555
OpenBCI_GUI: initSystem: -- Init 4 -- 701555
Hub: processConnect: made it -- 701834
Hub: setBoardType(): sending 'daisy -- 701834
Hub: processBoardType: success, starting SD card now -- 701835
Hub: sdCardStart(): sending '5min' with value 1
sd card setting set so now attempting to sync channel settings
New widget [11] selected for container...
Your cursor is in Networking
dy = 0
Acc Widget -- Screen Resized.
[... 37 lines cut ...]
openBCI_GUI: startButton was pressed...starting data transfer...
startRunning...
DEBUG: start data transfer
Cyton: startDataTransfer(): writing 'b' to the wifi shield...
Hub: sendCommand(char): sending 'b'
openBCI_GUI: stopButton was pressed...stopping data transfer...
OpenBCI_GUI: stopRunning: stop running...
Cyton: startDataTransfer(): writing 's' to the serial port...
Hub: sendCommand(char): sending 's'
openBCI_GUI: startButton was pressed...starting data transfer...
startRunning...
DEBUG: start data transfer
Cyton: startDataTransfer(): writing 'b' to the wifi shield...
Hub: sendCommand(char): sending 'b'
openBCI_GUI: stopButton was pressed...stopping data transfer...
OpenBCI_GUI: stopRunning: stop running...
Cyton: startDataTransfer(): writing 's' to the serial port...
Hub: sendCommand(char): sending 's'

**Powered board off and on**

Next 2 files failed 
SD Card was corrupted.

Also requires restart of OpenBCI GUI/Hub

msmadan, 2017-11-02:
@aj-ptw  I tested the cyton daisy combo with wifi shield with digital input randomly given in D11 and D17.worked great data saved via wifi.Refer attached data.Digital input widget worked perfectly .Thanks for all the help.
[OpenBCI-RAW-2017-11-02_19-32-53.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/1439696/OpenBCI-RAW-2017-11-02_19-32-53.txt)
Only one small problem i saw was after first set of  testing , the board had to be power cycled to get it working for the next set.

Thanks for the help

-Raja

aj-ptw, 2017-11-02:
What firmware are you running on the WiFi shield?

msmadan, 2017-11-02:
V1.1.3

aj-ptw, 2017-11-03:
You should update to 1.3.0

msmadan, 2017-11-03:
will do ,thanks
-Raja

gerrievanzyl, 2017-11-03:
@aj-ptw When you have to minutes, could you have a look on how to connect currentXvalue to the auxData marker stream so that the values come through.  Since you worked on all the data yesterday and today, I am sure it will take about 60 seconds. Thanks for all your great work!

msmadan, 2017-11-06:
@aj-ptw After updating the firmware on the wifishield to V1.3.0 the problem to need to power cycle has eliminated. Cycton+daisy+wifi shield combo works well, I am able to save digital trigger data with time stamp.
Thanks AJ for all your help
Raja

aj-ptw, 2017-11-06:
@msmadan great to hear, i'm about to release a new beta for v3.2.0

aj-ptw, 2017-11-06:
Could you please test the new beta v1.3.2 release on your systems to verify this issue is closed.

https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.2.0-beta2

msmadan, 2017-11-06:
will do
-Raja

msmadan, 2017-11-06:
I see there is zadig exe in the windows version , do I have to update the usb drivers?

aj-ptw, 2017-11-06:
Nope, just throw that in there for ganglion users.

msmadan, 2017-11-06:
ok

msmadan, 2017-11-11:
@aj-ptw ; checked the new version works great.
I just  got my second wifi shield , finished setting up today ,worked great.indiviually.
Quick question , how do i network both  the wifi shield (both  have 16 channel cyton) on the GUI so I can see both the data simulataneously , I tried the network widget but doesnot seem to work,May be I doing something wrong,Is there a write up for networking I can refer to .

Raja

aj-ptw, 2017-11-11:
Simultaneous WiFi shield to GUI data acquisition has not been demonstrated yet! You should open an issue if you want this feature and we can plan it out. You are not seeing a bug though, this is not possible with the current GUI!

msmadan, 2017-11-13:
OK thanks
-Raja

msmadan, 2017-11-13:
@aj-ptw , Did not know which section to ask this question. Please let me know if I need to move this to a different section

I have 2 cyton + daisy combination and 2 wifi shield

I think I have a problem with one of the cyton or daisy board connection with the wifi shield but I don't know if I am correct or what to look for.
Here is the problem :
When I connect the cyton Daisy combination to the GUI via Bluetooth I am getting data on all 16 channels, when I check the data file there is data stored on all the channels with accelerometer data and timestamp
When I  connect cyton + daisy to gui with wifishield  I am not getting data from the daisy . I know it is not the wifi shield or the daisy which has the problem since i checked it with the other cyton board it is giving data. I think it is a problem with some pin connection between cycton and the wifishield ,can you please let me know what you think. Also I am getting data from cycton board and accelerometer just not the daisy channels. If I remove the daisy I am getting data from cycton channels and accelerometer. 

Thanks
Raja

[2 more replies]

commit 2677da98 (2017-11-03): FIX: #272
commit 80a51bde (2017-11-03): ADD: Parsing data for board mode if stop byte is raw aux fixes #272
commit ab942ae2 (2017-11-03): FIX: #272 I can see the digital data, using the new digital widget outputting the digital data