OpenBCI/OpenBCI_GUI #277: Aux channels not saved in file when running 16 channel system

gerrievanzyl, 2017-11-01:
AJ,

This is a real bug.  When I remove the Daisy board AUX1 - AUX3 appears as they should in the OpenBCI text file.  As soon as I go to 16 channels all I get back in  AUX1 - AUX3  are zeros.  The same applies to the acceleration data in the DEFAULT board mode. 

This happens using BLE as well as wifi.  For wifi I have tested sample rates from 250 to 1000Hz and the bug is repeatable.  

OSX High Sierra
OpenBCI_GUI - Latest master 3.1.0 as well as release .app
Board firmware - Latest master + LED functions

**Experimental process**
_For Default (accel) mode_
1. Use downloaded release GUI 3.1.0
2. Connect to board (either 8 or 16 channels)
3. Set sampling rate
4. Start board 
5. Check .txt files

_For Marker data on the aux channel:_
1. Use custom GUI (based on latest github master with minor changes)
2. Send "/4" to put the board into maker mode
3. Start board
4. Send "`n" markers (n -> 0:127)
4. LED flash on board indicates markers are correctly received
5. Check .txt files

Without Daisy board (8 Channels) the markers are correctly recorded in the file as well as Acceleration info (only every 40 samples)  (BLE and Wifi)

With the Daisy board all the values for Aux/Accel in the files are Zero  (for both BLE and Wifi)

aj-ptw, 2017-11-01:
Sounds like a duplicate of #271

aj-ptw, 2017-11-01:
#277

gerrievanzyl, 2017-11-01:
AJ,

#271 is a feature request to add a board mode to the GUI.

#277 is bug where irrespective the board mode (BOARD_MODE_DEFAULT or BOARD_MODE_MARKER) none of the AUX information makes it back to the file.   

I have looked carefully at the firmware code and I can see no issue (except that the aux channels are sent through twice - once with the first 8 channels and again with the next 8 channels ).  I believe that the issue must be in the OpenBCI_GUI code.   

Does this make sense?

gerrievanzyl, 2017-11-01:
Maybe a duplicate of #272?

aj-ptw, 2017-11-01:
Possibly, just know there is effort to fix!! Tomorrow stay tuned

gerrievanzyl, 2017-11-01:
AJ, I may have some time tomorrow.  This is a super important issue for me and I will work with you to fix it - I just need guidance.  For the science fair project the girls need the wifi board saving 16 channels for their data gathering this weekend - or else we are in big trouble. 
Gerrie

aj-ptw, 2017-11-01:
@gerrievanzyl

I’m working on this first thing tomorrow. I had to address other fires. This has my full attention. Please wait to dive into till I finish the code tomorrow.

gerrievanzyl, 2017-11-01:
I never doubted that there is effort to fix.  Thanks for all the great work
you do!

I will have some time after 11am EST.  I may not be a Jedi Master, but I
can help (just don't ask me to look at JS code - all seems French to me!)

Gerrie

gerrievanzyl, 2017-11-01:
AJ, just confirming, you have not started working on this issue, or have you ?

Gerrie

aj-ptw, 2017-11-01:
I have not addressed this issue yet, now that i think about it, I have not tested with 16 channel. This is a duplicate of #272

gerrievanzyl, 2017-11-01:
Agree duplicate.  Should I close this one?

gerrievanzyl, 2017-11-02:
AJ,

I am tearing my hair out.

The new software I pulled from you was working moderately well (as you know
did not correctly write Aux to file for 16 channels) and then suddenly
stopped working.

I made no changes - I was just testing.

The really strange thing is that every time it fails it seems to do so at a
different place see below for extracts of the logs.  (BTW does the software
write more comprehensive logs somewhere - or is it just the console logs we
work with?)

Here are examples of the two failure modes:

*Example 1*
=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: initSystem: -- Init 0 -- 25963
timeOfInit = 25963
OpenBCI_GUI: initSystem: Preparing data variables...
OpenBCI_GUI: initSystem: -- Init 1 -- 25964
OpenBCI_GUI: initSystem: -- Init 2 -- 25965
OpenBCI_GUI: initSystem: -- Init 3 -- 25965
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
OpenBCI_GUI: stopRunning: stop running...
Closing any open SD file. Writing 'j' to OpenBCI.
disconnecting serial
Failed to connect to data source...
OpenBCI_GUI: initSystem: -- Init 4 -- 26067

*Example 2*
=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: initSystem: -- Init 0 -- 20969
timeOfInit = 20969
OpenBCI_GUI: initSystem: Preparing data variables...
OpenBCI_GUI: initSystem: -- Init 1 -- 20972
OpenBCI_GUI: initSystem: -- Init 2 -- 20977
OpenBCI_GUI: initSystem: -- Init 3 -- 20977
  3a -- 20977
  3b -- 20978
cyton: openNewLogFile: opened ODF output file:
SavedData/OpenBCI-RAW-2017-11-01_19-36-50.txt
ChannelController: createChannelSettingButtons: creating channel setting
buttons...
comPorts = [/dev/cu.Bluetooth-Incoming-Port, /dev/cu.MALS, /dev/cu.SOC,
/dev/cu.usbserial-DQ007Q8L, /dev/tty.Bluetooth-Incoming-Port,
/dev/tty.MALS, /dev/tty.SOC, /dev/tty.usbserial-DQ007Q8L]
accel w 1022
accel h 630
accel x 1
accel y 109
maker w 1022
marker h 630
marker x 1
marker y 109
widgets.size() = 13
widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
widgetOptions:
[Time Series, FFT Plot, Networking, Band Power, Accelerometer, Head Plot,
EMG, Focus Widget, Pulse Sensor, Digital Read, Analog Read, Marker Mode,
Widget Template 1]
Powering 3 widgets up, and remapping.
Activating widget [0]
Activating widget [1]
Activating widget [2]
  3c -- 22072
OpenBCI_GUI: initSystem: -- Init 4 -- 22072
Hub: processConnect: made it -- 22595
Hub: setBoardType(): sending 'cyton -- 22595
Hub: parseMessage: connect: success, starting SD card now -- 22595
Hub: sdCardStart(): sending '
Hub: processBoardType: success, starting SD card now -- 22596
Hub: sdCardStart(): sending '
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
OpenBCI_GUI: stopRunning: stop running...
Closing any open SD file. Writing 'j' to OpenBCI.
disconnecting serial

gerrievanzyl, 2017-11-02:
Now for the failures when I connect via wifi (the previous email was BLE)

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: initSystem: -- Init 0 -- 289905
timeOfInit = 289905
OpenBCI_GUI: initSystem: Preparing data variables...
OpenBCI_GUI: initSystem: -- Init 1 -- 289908
OpenBCI_GUI: initSystem: -- Init 2 -- 289914
OpenBCI_GUI: initSystem: -- Init 3 -- 289914
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
OpenBCI_GUI: stopRunning: stop running...
Closing any open SD file. Writing 'j' to OpenBCI.
Failed to connect to data source...
OpenBCI_GUI: initSystem: -- Init 4 -- 290015
Hub: processConnect: made it -- 291438
Hub: setBoardType(): sending 'cyton -- 291438
Hub: parseMessage: connect: success, starting SD card now -- 291438
Hub: sdCardStart(): sending '
Hub: processBoardType: success, starting SD card now -- 291439
Hub: sdCardStart(): sending '
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
OpenBCI_GUI: stopRunning: stop running...
Closing any open SD file. Writing 'j' to OpenBCI.

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: initSystem: -- Init 0 -- 42076
timeOfInit = 42076
OpenBCI_GUI: initSystem: Preparing data variables...
OpenBCI_GUI: initSystem: -- Init 1 -- 42079
OpenBCI_GUI: initSystem: -- Init 2 -- 42083
OpenBCI_GUI: initSystem: -- Init 3 -- 42083
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
OpenBCI_GUI: stopRunning: stop running...
Closing any open SD file. Writing 'j' to OpenBCI.
Failed to connect to data source...
OpenBCI_GUI: initSystem: -- Init 4 -- 42185
Hub: processConnect: made it -- 42581
Hub: setBoardType(): sending 'cyton -- 42581
Hub: parseMessage: connect: success, starting SD card now -- 42581
Hub: sdCardStart(): sending '5min
Hub: processBoardType: success, starting SD card now -- 42581
Hub: sdCardStart(): sending '5min
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
OpenBCI_GUI: stopRunning: stop running...
Closing any open SD file. Writing 'j' to OpenBCI.

gerrievanzyl, 2017-11-02:
How would you like me to pass on these issues to you.   Are you happy with
these posts, or would you like me to create new issues on github.

For example, I have found that when I try to exit the GUI by clicking on
SystemControl Panel it gives me only the option to start the system.

​

gerrievanzyl, 2017-11-02:
Ok - making some progress here.

I disabled the new digitalRead and analogRead widgets (but not the new
MarkerMode widget) and now I can get it to connect if.....

However, GUI will *not* connect unless:

   1. SD card is inserted
   2. Write to SD is enabled

If these conditions are met, then the board will connect with the widgets
disabled. (see comment out section below)

//only instantiate this widget if you are using a Cyton board for live
streaming
  if(eegDataSource != DATASOURCE_GANGLION){
    w_pulsesensor = new W_PulseSensor(_this);
    w_pulsesensor.setTitle("Pulse Sensor");
    addWidget(w_pulsesensor, w);
    // println("  setupWidgets pulse sensor -- " + millis());

 /*   w_digitalRead = new W_DigitalRead(_this);
    w_digitalRead.setTitle("Digital Read");
    addWidget(w_digitalRead, w);

    w_analogRead = new W_AnalogRead(_this);
    w_analogRead.setTitle("Analog Read");
    addWidget(w_analogRead, w);
*/
    w_markermode = new W_MarkerMode(_this);
    w_markermode.setTitle("Marker Mode");
    addWidget(w_markermode, w);

  }

  w_template1 = new W_template(_this);
  w_template1.setTitle("Widget Template 1");
  addWidget(w_template1, w);

gerrievanzyl, 2017-11-02:
How can I send pull request to you aj-ptw development branch?

oops, I tried following online instructions - but seems I messed up somehow
- pull rquest went through to OpenBCI/OpenBCI_GUI.git

Here is another clue:

I can now get the board to reliably connect as long as I reboot (power
up/power down) it after every session.

Hope all this feedback helps.

Gerrie

gerrievanzyl, 2017-11-02:
Closing this issue in favor of #272

closed 2017-11-02 by gerrievanzyl

tzvet, 2017-12-11:
@gerrievanzyl && @aj-ptw 

I just tried the release 3.2.0-rc2. I am interested by the markers correctly put in the SD card and noted few issues when testing the most recent release:

1. Using the widget from @gerrievanzyl ()W_Marker...) the SD card file does not contain the markers (i would like to have them; i also noted that @gerrievanzyl wrote somewhere that actually he had the markers correctly put in the SD card file but i think he was using the wifi shield). In my case it just puts zeros on the 3 AUX values.
2. The markers sent back with the "mean" data are not always present, i.e. sometimes they are missing; in one trial the "mean" data saved by the GUI contained only 4 out of the 8 markers i sent; in another only 6 out of 33 ...

Would it be easy to fix the SD card writing of the markers done by @gerrievanzyl ?

More general question:
If streaming is stopped, it means that data is not recorded on computer, but also on SD card !!! Is it possible to have one button (e.g. in W_Marker widget) for recording on SD card and sending markers without having the data sent to the computer ?
This question comes from my impression that streaming the data between cyton-dongle for observing on-the-fly makes an important hub problem and i think the markers are lost from time to time because of the limited bandwidth of my system (bluetooth dongle with cyton+daisy 16 channels). Or would you suggest another (simpler?) solution (e.g. python?)

appreciate any feedback

tzvet, 2017-12-12:
@aj-ptw : Thanks !!!! It works, All markers are in the SD card file.
@gerrievanzyl : thanks for the Marker widget ! But do you also have strong noise on EEG channels at marker onset ? (see further below).

I tested the timing, generally a delay of ~3-4ms, with std ~1ms (n=20 markers).

Just as a side note, the SD  card markers are not in the same format as the ones sent back to the computer: e.g. writing `9 gives back 57 on the computer, but it writes 0.007125 on the SD card... some conversion error, but still fine because different markers get different values :)

Now there is another quite some big problem, sending Markers makes a huge noise on ALL electrodes at marker onset. Any solution for this ? 

See screenshots below:
[image]
[image]

aj-ptw, 2017-12-12:
@tzvet that's great your getting the system to start working!

Can you please open two issues?

One for

> Just as a side note, the SD card markers are not in the same format as the ones sent back to the computer: e.g. writing `9 gives back 57 on the computer, but it writes 0.007125 on the SD card... some conversion error, but still fine because different markers get different values :)

and another for

> Now there is another quite some big problem, sending Markers makes a huge noise on ALL electrodes at marker onset. Any solution for this?

I do not want to keep commenting on a closed issue :) I love that you included screen shots that's awesome! Also if you could explain what you did to send the markers, I would love to know!

With two issues, we can work together to close each one, and maybe you, myself or another community member will write the code or patch! We are glad to help you develop your skills, no matter whatever level you are, we want to help!

commit 8f238f96 (2017-11-03): FIX: #277 add sdcardstart