OpenBCI/OpenBCI_GUI #278: SD Card writing is not working

gerrievanzyl, 2017-11-01:
Here is another issue I am experiencing.  I cannot save any data on the SD card.

Latest firmware (100% standard) from master
Latest release GUI

Tried:  
- 2 different SD cards
- Reformatting SD cards (with the SD Association Formatter)
- Wifi and BLE

gerrievanzyl, 2017-11-01:
I just looked at the console output of GUI 3.1.0 and could find no reference to SD Card init.

In particular, I remembered this line from prior versions:
  public void sdCardStart(int sdSetting) {
    String sdSettingStr = cyton.getSDSettingForSetting(sdSetting);
    println("Hub: sdCardStart(): sending \'" + sdSettingStr);
    write(TCP_CMD_SD + "," + TCP_ACTION_START + "," + sdSettingStr + TCP_STOP);
  }

There is no record of this "Hub: sdCardStart(): sending \'" in the logs:

Example:
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Mac
Welcome to the Processing-based OpenBCI GUI!
Last update: 9/5/2016
For more information about how to work with this code base, please visit: http://docs.openbci.com/OpenBCI%20Software/
Graphics & GUI Library: ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
attempting to load images for presentation...
DONE loading images!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
Instantiating hub object...
Hub: clientEvent: handshake complete
click
setInterface: curInterface: -1
setInterface: curInterface: 2
Setting sample rate for Cyton to 1000Hz

sample rate set to: 1000
Transfer Protocol set to wifi
refreshWifiList
click
SD setting = 1
click
ControlPanel — port is open: false
initSystem yoo

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: initSystem: -- Init 0 -- 24990
timeOfInit = 24990
OpenBCI_GUI: initSystem: Preparing data variables...
OpenBCI_GUI: initSystem: -- Init 1 -- 24995
OpenBCI_GUI: initSystem: -- Init 2 -- 25004
OpenBCI_GUI: initSystem: -- Init 3 -- 25004
  3a -- 25004
  3b -- 25004
cyton: openNewLogFile: opened ODF output file: SavedData/OpenBCI-RAW-2017-10-31_21-28-42.txt
ChannelController: createChannelSettingButtons: creating channel setting buttons...
comPorts = [/dev/cu.Bluetooth-Incoming-Port, /dev/cu.MALS, /dev/cu.SOC, /dev/cu.usbserial-DQ007Q8L, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.MALS, /dev/tty.SOC, /dev/tty.usbserial-DQ007Q8L]
accel w 1022
accel h 630
accel x 1
accel y 109
widgets.size() = 10
widgetOptions.size() = 10
wm.widgetOptions.size() = 10
wm.widgetOptions.size() = 10
wm.widgetOptions.size() = 10
wm.widgetOptions.size() = 10
wm.widgetOptions.size() = 10
wm.widgetOptions.size() = 10
wm.widgetOptions.size() = 10
wm.widgetOptions.size() = 10
wm.widgetOptions.size() = 10
wm.widgetOptions.size() = 10
widgetOptions:
[Time Series, FFT Plot, Networking, Band Power, Accelerometer, Head Plot, EMG, Focus Widget, Pulse Sensor, Widget Template 1]
Powering 3 widgets up, and remapping.
Activating widget [0]
Activating widget [1]
Activating widget [2]
Hub: processConnect: made it -- 25656
Hub: setBoardType(): sending 'cyton -- 25656
  3c -- 26096
OpenBCI_GUI: initSystem: -- Init 4 -- 26096
openBCI_GUI: startButton was pressed...starting data transfer...
startRunning...
DEBUG: start data transfer
BoardCyton: startDataTransfer(): Set board mode to 4 (Marker mode)
BoardCyton: startDataTransfer(): hub.sendCommand('/4') to the wifi shield ...
Hub: sendCommand(String): sending '/4'
Cyton: startDataTransfer(): writing 'b' to the wifi shield...
Hub: sendCommand(char): sending 'b'
Hub: apparent sampleIndex jump from Serial data: 204 to  223.  Keeping packet. (1)
numPacketsDropped = 19
Hub: apparent sampleIndex jump from Serial data: 2 to  90.  Keeping packet. (2)
numPacketsDropped = 88
Hub: apparent sampleIndex jump from Serial data: 213 to  232.  Keeping packet. (3)
numPacketsDropped = 19
Hub: apparent sampleIndex jump from Serial data: 11 to  114.  Keeping packet. (4)
numPacketsDropped = 103
Hub: apparent sampleIndex jump from Serial data: 246 to  9.  Keeping packet. (5)
numPacketsDropped = 18
Hub: apparent sampleIndex jump from

aj-ptw, 2017-11-01:
Is either card a Sandisk Class 10?

gerrievanzyl, 2017-11-01:
Yep - is that the issue?

gerrievanzyl, 2017-11-01:
Also just tried a Samsung 16GB - still no joy.  

It really does not look like this is ever called:

public void sdCardStart(int sdSetting) {
String sdSettingStr = cyton.getSDSettingForSetting(sdSetting);
println("Hub: sdCardStart(): sending '" + sdSettingStr);
write(TCP_CMD_SD + "," + TCP_ACTION_START + "," + sdSettingStr + TCP_STOP);
}

aj-ptw, 2017-11-01:
looks like a bug

gerrievanzyl, 2017-11-01:
AJ, I am ready, I have some time.  Anything I can help with?

aj-ptw, 2017-11-01:
Hey so on one of the returns from the hub we need to call sdCardStart if the sdSetting is greater than 0. It defaults to -1.

aj-ptw, 2017-11-01:
@gerrievanzyl please pull my `development` branch.

gerrievanzyl, 2017-11-01:
AJ, How do I do that?

Do I pull this to my branch or create new branch?

Gerrie

gerrievanzyl, 2017-11-01:
I and using the OSX Github Desktop app

aj-ptw, 2017-11-01:
Ok I never use the Github Desktop, only terminal, but many people do use the Desktop app.

That being said, I imagine you have custom code in your development branch that is not merged into my development branch, is that so?

gerrievanzyl, 2017-11-01:
Yes I have some of my own code in branch called feature/UDPmarkers

As I understand it I am trying to pull in the changes you have made to your
development branch.

What am I missing?

Gerrie-MacBook-Pro:OpenBCI_GUI gerrie$ git remote -v

OpenBCI https://github.com/OpenBCI/OpenBCI_GUI.git (fetch)

OpenBCI https://github.com/OpenBCI/OpenBCI_GUI.git (push)

origin https://github.com/gerrievanzyl/OpenBCI_GUI.git (fetch)

origin https://github.com/gerrievanzyl/OpenBCI_GUI.git (push)

Gerrie-MacBook-Pro:OpenBCI_GUI gerrie$ git branch -v

  development        9c545e0 Merge remote-tracking branch 'OpenBCI/master'
into development

* feature/UDPmarkers 6d40dbe fixed the Board mode error in the Hub

  master             5b51197 Merge pull request #274 from
OpenBCI/development

Gerrie-MacBook-Pro:OpenBCI_GUI gerrie$ git pull OpenBCI development

From https://github.com/OpenBCI/OpenBCI_GUI

 * branch            development -> FETCH_HEAD

Already up-to-date.

Gerrie-MacBook-Pro:OpenBCI_GUI gerrie$ git fetch OpenBCI

Gerrie-MacBook-Pro:OpenBCI_GUI gerrie$ git merge OpenBCI/development

Already up-to-date.

Gerrie-MacBook-Pro:OpenBCI_GUI gerrie$ git remote add ajptw
https://github.com/aj-ptw/OpenBCI_GUI.git

(py27) Gerrie-MacBook-Pro:OpenBCI_GUI gerrie$ git remote -v

OpenBCI https://github.com/OpenBCI/OpenBCI_GUI.git (fetch)

OpenBCI https://github.com/OpenBCI/OpenBCI_GUI.git (push)

ajptw https://github.com/aj-ptw/OpenBCI_GUI.git (fetch)

ajptw https://github.com/aj-ptw/OpenBCI_GUI.git (push)

origin https://github.com/gerrievanzyl/OpenBCI_GUI.git (fetch)

origin https://github.com/gerrievanzyl/OpenBCI_GUI.git (push)

git merge ajptw/development

merge: ajptw/development - not something we can merge

gerrievanzyl, 2017-11-01:
OK tried this.....

Has it worked?

Gerrie-MacBook-Pro:OpenBCI_GUI gerrie$ git fetch ajptw

remote: Counting objects: 222, done.

remote: Compressing objects: 100% (19/19), done.

remote: Total 222 (delta 174), reused 179 (delta 167), pack-reused 36

Receiving objects: 100% (222/222), 104.10 KiB | 1.13 MiB/s, done.

Resolving deltas: 100% (174/174), completed with 54 local objects.

From https://github.com/aj-ptw/OpenBCI_GUI

 * [new branch]      add-gitignore            -> ajptw/add-gitignore

 * [new branch]      bdf-plus-writer          -> ajptw/bdf-plus-writer

 * [new branch]      bug-split-line           -> ajptw/bug-split-line

 * [new branch]      bug-windows-cat-kill     -> ajptw/bug-windows-cat-kill

 * [new branch]      dev                      -> ajptw/dev

 * [new branch]      development              -> ajptw/development

 * [new branch]      enh-impedance            -> ajptw/enh-impedance

 * [new branch]      fix-112                  -> ajptw/fix-112

 * [new branch]      fix-145                  -> ajptw/fix-145

 * [new branch]      fix-151                  -> ajptw/fix-151

 * [new branch]      fix-bdf                  -> ajptw/fix-bdf

 * [new branch]      fixes                    -> ajptw/fixes

 * [new branch]      gabe-playback-networking ->
ajptw/gabe-playback-networking

 * [new branch]      gangion-freq-fix         -> ajptw/gangion-freq-fix

 * [new branch]      ganglion-wack            -> ajptw/ganglion-wack

 * [new branch]      high-level-fft           -> ajptw/high-level-fft

 * [new branch]      joel-pulse-sensor        -> ajptw/joel-pulse-sensor

 * [new branch]      master                   -> ajptw/master

 * [new branch]      wifi                     -> ajptw/wifi

 * [new branch]      wifi-update              -> ajptw/wifi-update

 * [new branch]      ws-proposal-branch       -> ajptw/ws-proposal-branch

(py27) Gerrie-MacBook-Pro:OpenBCI_GUI gerrie$ git merge ajptw/development

Auto-merging OpenBCI_GUI/W_PulseSensor.pde

Auto-merging OpenBCI_GUI/OpenBCI_GUI.pde

Auto-merging OpenBCI_GUI/InterfaceHub.pde

Auto-merging OpenBCI_GUI/DataLogging.pde

Auto-merging OpenBCI_GUI/BoardCyton.pde

Merge made by the 'recursive' strategy.

 CHANGELOG.md                    |  10 +++

 OpenBCI_GUI/BoardCyton.pde      |  16 ++++

 OpenBCI_GUI/DataLogging.pde     |  54 ++++++++++--

 OpenBCI_GUI/InterfaceHub.pde    |  47 +++++++---

 OpenBCI_GUI/InterfaceSerial.pde |   2 +-

 OpenBCI_GUI/OpenBCI_GUI.pde     |   2 +-

 OpenBCI_GUI/W_AnalogRead.pde    | 493
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

 OpenBCI_GUI/W_DigitalRead.pde   |  93 ++++++++++++++++++++

 OpenBCI_GUI/W_PulseSensor.pde   |   4 +-

 OpenBCI_GUI/WidgetManager.pde   |   9 ++

 10 files changed, 711 insertions(+), 19 deletions(-)

 create mode 100644 OpenBCI_GUI/W_AnalogRead.pde

 create mode 100644 OpenBCI_GUI/W_DigitalRead.pde

(py27) Gerrie-MacBook-Pro:OpenBCI_GUI gerrie$

gerrievanzyl, 2017-11-01:
AJ,

Seems something is wrong here.  The sdCardStart is only called if
eedDataSource != CYTON

       if (eegDataSource == DATASOURCE_CYTON) {
          if (nchan == 8) {
            setBoardType("cyton");
          } else {
            setBoardType("daisy");
          }
        } else {
          if (sdSetting >= 0) {
            println("Hub: parseMessage: connect: success, starting SD card
now -- " + millis());
            sdCardStart(sdSetting);
          } else {
            println("Hub: parseMessage: connect: success! -- " + millis());
            initAndShowGUI();
          }
        }

gerrievanzyl, 2017-11-01:
I removed the if conditionality and I see the SD card messages in the
console - still no joy.

No file on the SD card

BTW I have the wifi board connected - SD and wifi are not mutually
exclusive - right?

Gerrie

aj-ptw, 2017-11-01:
Hey okay sorry i should have put more links!
I keep my code in progress in [aj-ptw/development](https://github.com/aj-ptw/OpenBCI_GUI/tree/development) repo/branch. When my code is stable, I'll merge into [openbci/development](https://github.com/OpenBCI/OpenBCI_GUI/tree/development).

aj-ptw, 2017-11-01:
> Seems something is wrong here.  The sdCardStart is only called if
eedDataSource != CYTON

There is no SD support on ganglion because of physical hardware limitations.

gerrievanzyl, 2017-11-01:
AJ, That is not what you code says!

If the board *is* CYTON then the SD card start is *not* activated.

Anyway, I have tested it by removing the conditionality and it still does
not create a file on the SD card.

Gerrie

aj-ptw, 2017-11-01:
@gerrievanzyl great call. try the new code please.

gerrievanzyl, 2017-11-01:
AJ You have fixed the SD Write - Thanks and well done.

I am onto writing a W_MarkerMode widget.

I have a question - how can I get access to the dataPacket.auxValues in the
widget?  (or any other version of the auxValues)

Gerrie

gerrievanzyl, 2017-11-01:
Also, can you confirm that the

  public void processData(String msg) {

method in Hub will deal with the markers correctly?  An I correct that the
marker will be saved in datapacket.auxValues?

Gerrie

gerrievanzyl, 2017-11-01:
Let me be more specific, will the markers packet also have
*stopByte == 0xC1*
*?*

gerrievanzyl, 2017-11-01:
Never mind figured it out:

BOARD_MARKER_MODE will send packets with 0xC1 Stopbyte

Gerrie

aj-ptw, 2017-11-01:
@gerrievanzyl awesome, I'm onto other projects, will be back tomorrow to finalize this work. Please keep up the great work, im really excited to see your work and development. I'm around if you need me!

gerrievanzyl, 2017-11-01:
AJ,

I am busy looking at your widget code

   1. Does the Ganglion support ANALOG READ?
   2. Does the Ganglion support DIGITAL READ?
   3. Does the Ganglion have and accelerometer?

Gerrie

aj-ptw, 2017-11-01:
> 1. Does the Ganglion support ANALOG READ?
no see [OpenBCI_Ganglion_Library/issues/9](https://github.com/OpenBCI/OpenBCI_Ganglion_Library/issues/9)
> 2. Does the Ganglion support DIGITAL READ?
no see [OpenBCI_Ganglion_Library/issues/9](https://github.com/OpenBCI/OpenBCI_Ganglion_Library/issues/9)
> 3. Does the Ganglion have and accelerometer?
Yes.

gerrievanzyl, 2017-11-01:
I did some simple testing and the SD card is saving I will close this thread

Thanks AJ

closed 2017-11-01 by gerrievanzyl

gerrievanzyl, 2017-11-02:
AJ, solved this particular bug.  Can you take me through the process to send you a pull request to you aj-ptw repo?

aj-ptw, 2017-11-02:
Hey checkout this sweet gif i made!
[image]

gerrievanzyl, 2017-11-02:
Thanks - that worked - but I realized that you must have pulled in my stuff
already!

ANDRO90, 2018-06-14:
Hello @aj-ptw 

I'm getting back to this closed issue because I keep getting corrupted files saved onto the internal SD (samsung 32 GB) with Cyton + WiFi (last firmware and GUI, with and without daisy, and with any protocol and sampling rate).

I came across this thread while looking for an already similar issue and one of your sentences confused me, in particular at some point you said **"Wifi and SD card are mutually exclusive"**

Could you elaborate on that statement? I haven't found any reference to that, neither in the source code nor on the web.

Thanks!!

paquettejp, 2019-04-17:
> Hello @aj-ptw
> 
> I'm getting back to this closed issue because I keep getting corrupted files saved onto the internal SD (Samsung 32 GB) with Cyton + WiFi (last firmware and GUI, with and without daisy, and with any protocol and sampling rate).
> When I say "corrupted" I mean that after converting them to csv with the GUI I get just 7/8 samples (even if I set the recording to 5 mins, had no issues during the recording, and properly closed the session to close the file), and some of them are even not complete, i.e., only some channels have values.
> 
> I came across this thread while looking for an already similar issue and one of your sentences confused me, in particular at some point you said **"Wifi and SD card are mutually exclusive"**
> 
> Could you elaborate on that statement? I haven't found any reference to that, neither in the source code nor on the web.
> 
> Thanks!!

Hello, I am also curious about the phrase: "Wifi and SD are mutually exclusive".

I am currently using a Wifi Shield + Cyton and at the moment I am not able to save data to my SD card.

commit 25e06531 (2017-11-03): FIX: #278 thanks to gerrievanzyl
commit 75a6d4f6 (2017-11-03): FIX: Remove more ptins and fixes #278 sd card