OpenBCI/OpenBCI_GUI #279: When starting up the GUI sends multiple SD card start commands

gerrievanzyl, 2017-11-01:
GUI from development branch

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: initSystem: -- Init 0 -- 16676
timeOfInit = 16676
OpenBCI_GUI: initSystem: Preparing data variables...
OpenBCI_GUI: initSystem: -- Init 1 -- 16679
OpenBCI_GUI: initSystem: -- Init 2 -- 16684
OpenBCI_GUI: initSystem: -- Init 3 -- 16684
  3a -- 16684
  3b -- 16684
cyton: openNewLogFile: opened ODF output file: SavedData/OpenBCI-RAW-2017-11-01_19-16-06.txt
ChannelController: createChannelSettingButtons: creating channel setting buttons...
comPorts = [/dev/cu.Bluetooth-Incoming-Port, /dev/cu.MALS, /dev/cu.SOC, /dev/cu.usbserial-DQ007Q8L, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.MALS, /dev/tty.SOC, /dev/tty.usbserial-DQ007Q8L]
accel w 1022
accel h 630
accel x 1
accel y 109
maker w 1022
marker h 630
marker x 1
marker y 109
widgets.size() = 13
widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
widgetOptions:
[Time Series, FFT Plot, Networking, Band Power, Accelerometer, Head Plot, EMG, Focus Widget, Pulse Sensor, Digital Read, Analog Read, Marker Mode, Widget Template 1]
Powering 3 widgets up, and remapping.
Activating widget [0]
Activating widget [1]
Activating widget [2]
  3c -- 17781
OpenBCI_GUI: initSystem: -- Init 4 -- 17781
Hub: processConnect: made it -- 18327
Hub: setBoardType(): sending 'cyton -- 18328
Hub: parseMessage: connect: success, starting SD card now -- 18328
Hub: sdCardStart(): sending '
Hub: processBoardType: success, starting SD card now -- 18328
Hub: sdCardStart(): sending '
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
OpenBCI_GUI: stopRunning: stop running...
Closing any open SD file. Writing 'j' to OpenBCI.
disconnecting serial

aj-ptw, 2017-11-01:
Are you sure you merged in the changes correct? This should only be called once. Do a Search for the function, it should be in there only three times if I remember correctly.

gerrievanzyl, 2017-11-02:
I found two calls for the SD card in succesive functions in

private void processBoardType(String msg) {

and

private void processConnect(String msg) {

I commented out the one second method:

        if (sdSetting >= 0) {
          println("Hub: parseMessage: connect: success, starting SD card
now -- " + millis());
          sdCardStart(sdSetting);
        } else {
          println("Hub: parseMessage: connect: success! -- " + millis());
          initAndShowGUI();
        }

gerrievanzyl, 2017-11-02:
Fixed and tested on my branch features/UDPmarkers

closed 2017-11-02 by gerrievanzyl