OpenBCI/OpenBCI_GUI #285: GUI Control Panel is too long for display on smaller screens

gerrievanzyl, 2017-11-06:
The new version of OpenBCI GUI with the Wifi settings that include Sample Rate and Latency is too long to display on the screens of MacBook Pro 13" computers. There is no way to access the Sample Rate and Latency buttons.

A possible solution would be to break out horizontally with these options

aj-ptw, 2017-11-06:
This sounds like a bad bug

Could you send me the exact line? [How to link to a specific line](https://stackoverflow.com/posts/31282559/revisions)

Thanks!

aj-ptw, 2017-11-12:
Can you please add a screen shot?

aj-ptw, 2017-12-05:
I fixed this with [v3.2.0-rc1](https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.2.0-rc2)

aj-ptw, 2017-12-13:
@gerrievanzyl please let me know if the [latest stable gui release](https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.2.0)

closed 2017-12-13 by aj-ptw