OpenBCI/OpenBCI_GUI #296: Markers on SD card do not have same format as the ones sent back to computer

tzvet, 2017-12-13:
This comes from #277  after suggestion of @aj-ptw 

## Problem
    Just as a side note, the SD card markers are not in the same format as the ones sent back to the computer: e.g. writing `9 gives back 57 on the computer, but it writes 0.007125 on the SD card... some conversion error, but still fine because different markers get different values :)

## Expected : looks like simple conversion issue on SD card
## Operating System: Lubuntu 16.04
## GUI Version: openBCI_GUI 3.2.0-latest release
## Running standalone app
## Type of OpenBCI board: cyton+daisy 16 channels eeg with standard dongle

aj-ptw, 2017-12-13:
Hi can you please explain how to use the marker widget? @gerrievanzyl added this widget

tzvet, 2017-12-14:
For the Markre Widget, I just pressed the only button present:
- when not in Marker mode it says "Set Marker On"
- when in Marker mode it says "Set Marker Off"

To me it looks like sending the Markers must be done outside the widget, which i did (see post #297)

tzvet, 2017-12-18:
UPDATE: i succeed to have an external time marker (photodiode) as input to D11. Made a test and all works fine, out of the problem of conversion; I used the Analog widget and recorded on the SD card. The values recorded varied between ~0.01 up to ~0.12 ; somehow the same order of magnitude as the values recorded when sending the computer generated Markers as described in #297 .

closed 2020-06-02 by retiutut