OpenBCI/OpenBCI_GUI #298: OpenBCIHub Fatal Error

tzvet, 2017-12-13:
I saw a reproducible error, though a bit randomly appearing.

When using the Marker widget, after few On/Off presses for Marker switching on/off a small window appears with this text:
> OpenBCIHub Fatal Error
> Cannot read property 'length' of undefined

### Operating System
lubuntu 16.04 + openBCI GUI 3.2.0 standalone

### Type of OpenBCI board
cyton + daisy 16 channels

aj-ptw, 2017-12-13:
@tzvet thanks so much for the bug report!

Could you try this with the accel mode marker?

Thanks!!

tzvet, 2017-12-14:
i cannot try this from the accelerometer widget because in the GUI's widget there is no button ...

aj-ptw, 2017-12-14:
Shoot I meant to say analog read, I wonder what firmware version your Cyton has.

tzvet, 2017-12-14:
Yep, the Analog widget also makes it... and i tried the Digital widget, it also gives this error...

aj-ptw, 2017-12-14:
Great. You should up date your Cyton v3.1.0

aj-ptw, 2017-12-14:
https://github.com/OpenBCI/OpenBCI_32bit_Library/releases/tag/v3.1.0

aj-ptw, 2017-12-14:
This sounds like a big, I’ll look at it shortly.

tzvet, 2017-12-14:
ok, about updating to v3.1.0, which docs/instructions should i follow ?
these ones : http://docs.openbci.com/Hardware/05-Cyton_Board_Programming_Tutorial#cyton-board-programming-tutorial-firmware-versions-2xx-3xx-fall-2016-present-install-firmware-from-arduino-library-manager-easiest

aj-ptw, 2017-12-14:
Yes.

aj-ptw, 2017-12-14:
I can't duplicate with v3.1.0 and current hub, how did the upgrade go?

tzvet, 2017-12-15:
So, i updated (it took the time it needed...). Result, the problem is still there. I tried the Marker and Analog  and Digital widgets, all have it, though it took longer time in 2 out of 3 cases to arrive at it. When the system crashes then when i press "Stop Data Stream" it also reports, on the terminal, smthg like this:
> java.lang.NullPointerException
>           at processing.net.Client.write(Unknown Source)
>           at processing.net.Client.write(Unknown Source)
>           at OpenBCI_GUI$Hub.write(OpenBCI_GUI.java:13391)
>           at OpenBCI_GUI$Hub.sendCommand(OpenBCI_GUI.java:12782)
>           at OpenBCI_GUI$Cyton.write(OpenBCI_GUI.java:1673)
>           at OpenBCI_GUI$Hub.stopDataTransfer(OpenBCI_GUI.java:1732)

etc.

Could you explain why the app under linux should be run as administrator ? I saw that somewhere on the main openbci web or in the forum, i also tried without sudo and it works. At least i did not find any difference between the two ways to run it.

And the problem is still present when running in administrator mode. i tried it.

tzvet, 2018-03-29:
Hello ! Sorry but i have not used the system for about 2 months... i have not updated the app since then. At the moment have no time to recheck it (uploading new app and other issues to fix), may be in about 2 weeks.

daniellasry, 2019-01-22:
Seems fixed as of GUI v4

closed 2019-01-22 by daniellasry