OpenBCI/OpenBCI_GUI #299: Ganglion BLE Connect MAC High Sierra

wyojustin, 2017-12-17:
## Problem

When I click
System Control Panel --> Live (From Ganglion) --> BLE (on Win from dongle)

in the GUI I see the ganglion device named "Ganglion-15c"

I click the device name and "start system".  Then the GUI hangs. 

## Expected 

I expect the GUI will display a play button.

## Operating System and Version

macOS High Sierra

## GUI Version
3.2

## Running standalone app

I am running the downloaded app

## Type of OpenBCI Board
Ganglion

Are you using a WiFi Shield?
No

aj-ptw, 2017-12-17:
The name is “Ganglion-15c”?? There should be four numbers/letters at the end and right away that seems troubling.

wyojustin, 2017-12-17:
Correct.  I missed the last character.

Ganglion-15c0

aj-ptw, 2017-12-17:
Okay great, thanks so much for the clarification.

Does any message get displayed on the bottom of the GUI? 

What is the exact version number of your OS?

wyojustin, 2017-12-17:
Message on bottom of screen: "Ganglion Device Name = Ganglion-15c0"
When it starts up it says "3.2.0"

aj-ptw, 2017-12-17:
> What is the exact version number of your OS?

Could you [follow this guide](https://support.apple.com/en-us/HT201260) to get your macOS version number?

wyojustin, 2017-12-17:
OSX 10.13.2 High Sierra

trevyn, 2017-12-27:
I also experienced this problem on macOS 10.12.6 -- Ganglion appears in the BLE device list, but "Start System" hangs the GUI.

Selecting "BDF+" instead of "OpenBCI" for the data log file format works and allows streaming. Then, clicking "Stop System" under "System Control Panel" causes a hang.

aj-ptw, 2017-12-27:
@trevyn Did you put the application in your 'Applications' folder?

trevyn, 2017-12-28:
Ah yes, moving to Applications resolves the issue.

closed 2018-03-04 by aj-ptw