OpenBCI/OpenBCI_GUI #302: 3.2.0 - Interface freezing when converting from SD format

GeneralFox2, 2017-12-26:
## Problem
I upgraded to 3.2.0 (published in Dec 2017) as I wanted to run the latest software.  I noticed that saving to SD card provides higher sample data so I am now using an SD card to capture the data stream on my Cryton and daughter board.

I copy the file from the SD Card to my computer.  I use the interface with option "Convert SD file for playback" because I want to be able to open the file using MATLAB.  The interface lets me make all my selections and then freezes after I select the file.  The file is not big - it's only 1 minute long.  After 1 hour I give up and end the program, reboot, and tried again.  

I checked the tutorial and noticed that I had not applied the "Run As Admin" in the Compatibility Mode for the programs - and set them appropriately.  This did not fix the issue.  I am willing to do a remote session if that helps (goto meeting, etc.).  

## Expected 

I expect the file to be converted as outlined in the tutorial on the Internet.  I expect to find the file in the folder outlined below:
C:\Users\Don\Documents\Processing\OpenBCI_GUI\data\EEG_Data

I do not wish to use the "record to PC" option any longer as I wish to have a better sampling rate.

## Operating System and Version

Windows 10 (always running the latest patches as forced by Microsoft)

## GUI Version

3.2.0 as published in December 2017

## Running standalone app

I am running the latest version as found here:
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.2.0

## Type of OpenBCI Board

Cyton+Daisy

Are you using a WiFi Shield?  NO  I am using the option to write to the SD card.  This has worked before but does not work any longer in this new version.  

[image]

aj-ptw, 2018-01-09:
Hey @GeneralFox2 sorry for the delay, was away for the holidays :) 

Looking into these issues now! Thanks for your support as always and I liked your matlab write up and shared it on twitter!

aj-ptw, 2018-01-09:
> I expect the file to be converted as outlined in the tutorial on the Internet. I expect to find the file in the folder outlined below:
C:\Users\Don\Documents\Processing\OpenBCI_GUI\data\EEG_Data

Could you post a link to this tutorial?

GeneralFox2, 2018-01-12:
> Hey @GeneralFox2 [1] sorry for the delay, was away for the holidays :)

Hi! 

Hope you had a great holiday season! 

Hey cool!  Thanks for sharing that on twitter! 

Don 

--------------------------------------- 

 

Links:
------
[1] https://github.com/generalfox2
[2]
https://github.com/OpenBCI/OpenBCI_GUI/issues/302#issuecomment-356361646
[3]
https://github.com/

GeneralFox2, 2018-01-12:
Hi, 

I see it has been updated - but this is still not the correct location.
On my computer the specified folder empty. 

In fact, the tutorial specifies the Processing program which is odd
considering that the program you need to run is the OpenBCI GUI - which
is even the GUI pictured. 

"Navigate to the OpenBCI generated file, and click OPEN. The GUI will
convert the file you selected. The resulting file will be placed inside
a file called EEG_Data in the Processing OpenBCI GUI Sketch Folder." 

Here is the link to the above text, in the tutorial.  The text can be
found at the bottom of the page: 

http://docs.openbci.com/Tutorials/09-Using_SD_Card_with_OpenBCI#using-sd-card-with-openbci-openbci-formatting-converting-hex-files-in-openbci-giu

Thanks, 

Don

----------------------------------------------------- 

 

Links:
------
[1]
https://github.com/OpenBCI/OpenBCI_GUI/issues/302#issuecomment-356362255
[2]
https://github.com/

aj-ptw, 2018-01-12:
>  Here is the link to the above text, in the tutorial. The text can be found at the bottom of the page:

Thanks, I will update the tutorial. I really appreciate your help and time!

aj-ptw, 2018-01-19:
@GeneralFox2 could you share the SD file? Are you sure anything is actually being written? I'm not sure how you are streaming over 250Hz without the wifi shield.

GeneralFox2, 2018-01-21:
Hi,

But I'm not streaming to the computer. When you select to record to the SD Card the unit still shows the averaged samples in the OpenBCI GUI. Once you are finished, you must then copy the file from the SD Card to the computer, and then use the OpenBCI GUI to convert it to a CSV file. 

I just finished running through the entire sequence again, starting completely from scratch (a new recording - 4 minutes long). I have furthered my signal conditioning to include removing from the beginning and end of all of the signals. I also published my EEG recording files used throughout the process.

You can see the updated post [here](http://generalfox.com/EEG_Research.htm).

Thanks,
Don

GeneralFox2, 2018-06-13:
Hi,

The latest version (3.3.7) - just downloaded yesterday (June 12, 2018) now works when extracting the file as expected.

However I have a new issue that I will post as a new issue.

Thanks,
Don

closed 2018-06-13 by GeneralFox2