OpenBCI/OpenBCI_GUI #305: Clean up the UDP marker listener in the OpenBCI_GUI

gerrievanzyl, 2018-01-11:
## Problem

1. Currently the UDP marker listener is started in the main GUI setup function.  This means that it is always running an not only started when the Marker mode is used.  This is not ideal.  
1. The current implementation hardwires the address of the incoming packets to 127.0.0.1 and port 51000.  This means the stimulus presentation software must run on the same machine as the GUI. 

## Expected 

1.  The Marker widget would be modified to include UDP address and port fields.  
1.  The UDP listener would be started only when maker mode is enabled  

## Operating System and Version

macOS/Windows/Linux

## GUI Version

3..1.0 onwards

## Type of OpenBCI Board

Cyton/Cyton+Daisy/Ganglion

## Are you using a WiFi Shield?

This works with bluetooth and wifi

aj-ptw, 2018-04-02:
Any progress with this?

retiutut, 2019-10-21:
The UDPMarker listener is supposed to listen for a marker from an external application and then the GUI sends a marker message, for example `9, to the Cyton board while in Marker Mode.

closed 2020-01-22 by retiutut