OpenBCI/OpenBCI_GUI #306: No data streaming in OpenBCI GUI

faheemersh, 2018-01-11:
## Problem

We have the dongle and cyton 8 channel/32 bit board connecting, but no data streams, In other words, all the channels show 0 uVrms. I have tried pressing the RST button and no luck. I did also press the program button once, but I don't think that had any affect on what's going on now. The system was working yesterday at 4:45 pm on my Windows computer. We were trying to get it to work with another computer (Mac) after that, but that computer had the same problem. 30 minutes later, I connect the system back into my computer, and I'm having the problem.

## Expected 
Should be seeing the data stream in the GUI. 

## Operating System and Version

Windows 10
## GUI Version

3.2
## Running standalone app

I am running the standalone app.
## Type of OpenBCI Board

Cyton 8 channel, 32 bit board
## Are you using a WiFi Shield?

No

aj-ptw, 2018-01-11:
Could it be a low battery?

faheemersh, 2018-01-11:
So I fully charged the battery yesterday. Somehow I got it to work again. I tried making sure that a successful connection was established by clicking the > arrow on the serial/com port. I hit status, which it the program showed that it failed to connect. Then I hit autoscan and it successfully connected. I hit status again to verify and started the system and then data stream. It's working again. I don't know if this is an official resolution of if something happened by chance. I just hope the system isn't inconsistent like this.

So I guess the question is at what point does the program grant access to the data stream screen? Is it when the dongle is successfully detected? Or when both the dongle and board have a connection?

aj-ptw, 2018-01-11:
@faheemersh you should **never** have to change the channel of the radio and dongle unless there is another Cyton running at the same time.

Now that your radio link has been fixed, and your battery charged, you will never have an issue again. If you are able to connect to the device by hitting "Start System" but no data comes in, that's most likely because the battery dies.

faheemersh, 2018-01-11:
I wasn't changing the channel, I knew that having another Cyton was only reason I would need to. I only when to the option to check the status of the connection. It is odd that it was working before, stopped working (even with a full battery) and then after doing what I did in the program, it started working again.

I came to the same conclusion yesterday, which is why I charged the battery. I got worried when it took over 4 hours to get the green LED to show up on the lipo USB charger. Is it okay to hook the USB charger into a wall adapter, so that it might charge faster? I don't want to damage the charger, which is why I'm asking.

aj-ptw, 2018-01-11:
> I only went to the option to check the status of the connection.

Awesome! :) I do that same!

> It is odd that it was working before, stopped working (even with a full battery) and then after doing what I did in the program, it started working again.

I agree it's very odd, please keep an eye on it and post here if you have to repeat.

> I came to the same conclusion yesterday, which is why I charged the battery.

This is a problem I want to crush with #304 but i didnt realize this would also happen with Cyton!!! I will update that issue to also include cyton, maybe even Ganglion. 

>  I got worried when it took over 4 hours to get the green LED to show up on the lipo USB charger.

What LiPo charger are you using? Could you please paste a link? I can then look at the charge rate and see if you can purchase a faster batter charger. At this point in my life, I always keep another LiPo battery charged and with me.

> I don't want to damage the charger, which is why I'm asking.

I'm going to assume your LiPo charge is limiting you to 500 mAh charge rate, and that plugging into the wall will not change the charge time. Also I plug mine into the wall all the time so have no fear.

faheemersh, 2018-01-11:
So this is actually a senior design project. One of my group members bought this charger from adafruit:  https://www.adafruit.com/product/1304 and this battery from sparkfun: https://www.sparkfun.com/products/13851

Also, I'm trying to stream data into MATLAB/BCILAB. I figured that I should just follow the LSL setup for OpenBCI with python and then try using BCILAB in MATLAB to grab the stream. I'm not sure if this is the correct step to take to stream the data into MATLAB. I also spoke with wjcroft (William) over on OpenBCI's website and he suggested that there is a way to get data directly into MATLAB from a serial port - I want to be able to get data (sampled at 250 Hz) into MATLAB to minimize delay. Do you have any suggestions?

aj-ptw, 2018-04-04:
Was this resolved?

faheemersh, 2018-04-04:
I apologize, yes, it was resolved!

closed 2018-04-04 by aj-ptw

Bpotapov, 2023-07-31:
Hi, could you help me. I think I see the same problem. My board works and doungle works too. Battery full. But  "Data streamining error". When I run autoscan I see error Error readining from serial port or sometime 060110. But I can start session. I when I do it I see this ... during session 5 sec and each chenal has text "Data from the board doesn't change"  
[image]