OpenBCI/OpenBCI_GUI #307: Array out of bounds/EEG Processing Error

drjcode, 2018-01-11:
## Problem

Converted SD card data using the processing GUI.  When trying to load the resulting converted data for playback by the GUI - getting an Array out of bounds/EEG Processing Error

I am including both the raw data file and the converted file

[OBCI_09.TXT](https://github.com/OpenBCI/OpenBCI_GUI/files/1624103/OBCI_09.TXT)

[SDconverted-2018-01-10_19-35-57.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/1624115/SDconverted-2018-01-10_19-35-57.txt)

Explain the problem

## Expected 

See playback of the SD card data in the GUI.

## Operating System and Version

Windows 10 Pro

## GUI Version

December 2017 release

## Running standalone app

Processing 3

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?
No

aj-ptw, 2018-01-12:
This is not a feature of the GUI and needs to be made. I think the SD card converter could use the same header as the OpenBCI output file format and then this feature would work!

Would you be interested in coding it up?

aj-ptw, 2018-04-02:
WIll need to add some info to the converted data file such as 

* Sample rate
* Number of channels

We could ask for this data when the file is first converted so the header is populated.

GeneralFox2, 2018-06-13:
Hi,

I would second the need for this.  I have no way of comparing the output of my MATLAB against the GUI because it won't play these recordings.

Thanks,
Don

ahsanic, 2018-07-31:
I think I have the same issue, this is the error in OPENBCI GUI latest versions, It doesnt play playback file.

retiutut, 2019-04-07:
@daniellasry This issue can be closed.

closed 2019-04-07 by daniellasry

GeneralFox2, 2019-04-18:
> Closed #307 [1].

Hi, 

Does this mean that the feature was implemented? 

Thanks,
Don

----------------------------------------------------- 

 

Links:
------
[1] https://github.com/OpenBCI/OpenBCI_GUI/issues/307
[2] https://github.com/OpenBCI/OpenBCI_GUI/issues/307#event-2257708101
[3]
https://github.com/

retiutut, 2019-04-19:
Yes.