OpenBCI/OpenBCI_GUI #308: Using External input in pin D11 as time Marker creates noise in the SD card recordings

tzvet, 2018-01-17:
## Problem

I tested an external time stamp as a Marker by using as input signal in pin D11 a simple Resistor-Diode design and the three pins GND-VDD-D11. A photosensor was put in front of a monitor with a simple light pulse appearing every 1 second for 10/144~0.070 seconds.
Recording was carried with Analog Widget, thus the 3 AUX channels are the pins D11-D13 (if i am not mistaken...it was ~2weeks ago) with a Cyton+Daisy 16 channels.

The markIV was simply laying on the computer minitower (therefore the strong noise in the signals, see snapshots), thus there is no other "signal" than the photosensor that should be on the D11 pin and all surrounding EM noise.

Analysing the SD card recorded file, i saw the square window on AUX channel 1, exactly what i wanted.
BUT
i also saw:
(1) AUX2 has also some effects, ~25% of the aplitude of the AUX1 channel
(2) all 16 channels have systematic and strong signals directly related to the D11 input ! The snapshot of the self processed 16 channels (notch filter + butterworth) show the 16 channels and the 2 AUX channels (these last two are not filtered, they are raw SD values but scaled times 1000 - see #296 )

## Expected 

to see the photodiode signal on D11, smthg like a square window of duration ~70ms exactly as it is already, BUT irrelevant noise on the other channels without artifacts due to input on D11.

## Operating System and Version

Linux ubuntu 16.04

## GUI Version

>= 3.2.0

## Running standalone app

Standalone

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

No

biomurph, 2018-01-17:
@tzvet This is quite a problem. First, I would say that seeing an artifact on one of the neighboring analog pins is not a surprise or unusual. The effect on the data is strong, but as you say the headset is 'sitting' on the table.

Can you share the exact circuit and connections you are making? I can't quite make out your setup from the photo.

tzvet, 2018-01-18:
## the circuit

[image]

aj-ptw, 2018-04-02:
@biomurph ping

leevonk, 2018-10-29:
I'm having this same problem. This is a Huuge bug. The whole point of recording markers in EEG data is to be able to subsequently do a peri-stimulus time histogram on the dataset to see if a post-stimulus effect can be observed in the EEG data. So any post-marker noise that appears on all channels is totally unacceptable and makes the markers pointless.

tzvet, 2018-10-30:
@leevonk : sorry to hear that you have the same problem. I decided to use the cyton + dongle wireless communication to send markers (see #297 , the problem noted there was not present on the data recorded on the SD card on the board, it is present only on the data received by the computer - and computer recorded data is not the raw data... well unless one uses the wifi, which seems to solve the pb).

I do not know if you can find another way to send non-electric markers to your board.

Some ideas i had at the time when i encountered the pb, but had no time to look at it simply because i had another simpler solution: (1) see if the amplitude of the electric marker can be decreased (it will make the effect much weaker); (2) find out the exact effect of your marker on the other electrodes and see if you can "subtract" it (your marker should be always the same - thus the effect is constant).
I am sorry, these are not perfect solutions but might help to go on with your system.

tzvet, 2019-08-29:
@wjcroft Thx! I would try the solution with >1000x higher resistor when take time to retest an external trigger. This definitely might help.

I also thought about separating the trigger voltage, because i used simply the VDD and GND levels given by the cyton/daisy pins. (though see @leevonk 's comment).

I leave this thread open until i have not tested the much higher resistor.

But OpenBCI managers can close it.
There is enough infos for any other user to know what's going on.

Thanks for the reply and keeping working on it !

retiutut, 2019-11-08:
I successfully tested the [OptoIsolator solution](https://docs.openbci.com/docs/02Cyton/CytonExternal#optoisolation) using a head-mounted RaspberryPi Zero W in the last month. 🔥 

- Artifacts will show if a channel does not have a quality connection. I recommend turning off unused channels.

- I am comfortable with making OptoIsolator the best solution for sending external triggers (since it looks like Marker Mode is only partially implemented in the GUI).

[image]
[image]

closed 2019-11-08 by retiutut

wjcroft, 2019-11-08:
Note that @tzvet WAS using an optoisolator, but had too much current flowing because of the 1K resistor value choice. It would still be helpful to hear from him if his issue was resolved using the suggestions previously mentioned.

retiutut, 2019-11-08:
Here is what I used:

[image]