OpenBCI/OpenBCI_GUI #309: Processing IDE won't compile, major list of errors.

shadowphile, 2018-01-21:
## Problem
I've loaded 3.3.6 Processing and successfully opened the sketch for the OpenBCI_GUI.  The Hub is running fine although the black cat is not in my tool bar (it was in the past so not sure what has changed).  
The standalone GUI opens fine and talks to my headset no problem.
## Expected 
I would like to just click Play and have the GUI come up.
## Operating System and Version
Win 7 64 SP1
## GUI Version
3.0
Are you running the downloaded app or are you running from Processing 3
Downloaded app ran fine (in the past). The IDE setup will not.
## Type of OpenBCI Board
Home-built, based on exact copy of Cyton (8-channel).  It worked great with the GUI.  I need to write my own widget and that is where my problem exists.
[image]

## Are you using a WiFi Shield?

No

aj-ptw, 2018-02-13:
Did you modify the files at all?

Did you drag and drop the library's into your libraries folder specified by preferences?

shadowphile, 2018-02-13:
thanks for the reply aj-ptw.
I did not modify anything.
It's many weeks so I don't remember about the library folder you mention.  I will look at let you know.

aj-ptw, 2018-04-02:
This issue will be resolved upon the release of the next GUI, v3.3.0 by removing all of the HUB constants throwing errors out of the hub object, it was a good idea to make this happen!

You can verify this is fixed in the `development` branch

closed 2018-04-02 by aj-ptw