OpenBCI/OpenBCI_GUI #310: Sub-options does not show when clicking "LIVE (from Cyton)"

rcsofficial, 2018-01-26:
When I click the LIVE (from Cyton) option, it doesn't do anything. The error message below says I should open the OpnBCI Hub prior to launch of OpenBCI GUI despite the fact that OpenBCI Hub is actually running before I ran the GUI. I did run the Hub as an administrator yet nothing happens.

[image]

aj-ptw, 2018-01-26:
Maybe Labview is using the same port number as the OpenBCI Hub. Does any error come up when the hub is launched?

rcsofficial, 2018-01-26:
No errors show up when the hub is launched. I tried uninstalling the
LabView but the GUI won't allow me to click the "Live from Cyton" anymore.
Prior to the LabView install, everything in the GUI worked fine for my
Windows 10 laptop.

Yours truly,

*Richmond Sim*
Academic: richmond.sim@obf.ateneo.edu
Personal: sim.richmond@gmail.com
General Mobile: +63 915 971 8434 / +63 929 723 4504

aj-ptw, 2018-01-26:
Do you think you could try running the GUI from processing? This will print out the errors the GUI is running into.

Sorry for these troubles!

rcsofficial, 2018-01-27:
I did try to run it using Processing and after a couple of library fixes
required, the GUI still did not run. I will respond back to check the exact
code in the console after I get back to my work station

rcsofficial, 2018-01-30:
I followed all instructions in your documentations and tried three
different computers. I am stuck here:

[image: Inline image 1]

The play button is green but the GUI does not appear. I've already waited
30 minutes to no avail.

Yours truly,

*Richmond Sim*
Academic: richmond.sim@obf.ateneo.edu
Personal: sim.richmond@gmail.com
General Mobile: +63 915 971 8434 / +63 929 723 4504

SoatOfficial, 2018-02-07:
I have the same problem, but when i click on " LIVE (from cyton) " the console show me "click"

aj-ptw, 2018-02-07:
What version?

SoatOfficial, 2018-02-08:
[image]

aj-ptw, 2018-02-08:
Is there an OpenBCIHub in your _data_ folder?

SoatOfficial, 2018-02-08:
No there isn't

[image]

aj-ptw, 2018-02-13:
You just need to drag and drop the [openbci hub](https://github.com/OpenBCI/OpenBCI_Hub)

Latest release is v1.3.9
https://github.com/OpenBCI/OpenBCI_Hub/releases/tag/v1.3.9

aj-ptw, 2018-03-13:
I believe i have solved these issues, please let me know if there is anything else needed.!

closed 2018-03-13 by aj-ptw