OpenBCI/OpenBCI_GUI #313: Sending external triggers through the lab streaming layer

faheemersh, 2018-02-16:
## Problem

I'm currently able to use an external trigger to send a marker in my OpenBCI EEG data. However, when trying to use this with the lab streaming layer, it doesn't work.

## Expected 

The marker from my P300 paradigm would be sent over the stream as a separate channel of input, either through vis_stream or ReceiveData.m from the lab streaming layer, or even to BCILAB.
## Operating System and Version

Windows 10

## GUI Version

v3.2.0

## Running standalone app

Running through processing as the standalone GUI freezes when trying to start the LSL stream.

## Type of OpenBCI Board

Cyton 32-bit board 8 channel

## Are you using a WiFi Shield?

No

aj-ptw, 2018-03-13:
Okay great thanks for the feature request!

Any idea what format the LSL expects this data as? Currently i believe we are just publishing an array of EEG data so I need a bit more info on how LSL wants this marker.

faheemersh, 2018-03-13:
So I was able to get the marker streams to work. The issue with LSL and what OpenBCI GUI seems to output is the number of columns. The format LSL expects (with markers) I believe is like the following:

columns 1-8: EEG data
column 9: marker information
column 10: timestamps

The reason I think this is because I am currently using OpenViBE for my senior design project and can successfully record data from the Cyton 32 bit board as well as the markers from the built-in Graz stimulation. The marker stream is always zero except when an event occurs. Each event has it's own ID. For example:

Start trial:  768
Black screen with cross: 786
Black screen with cross with cue: 769 (Left) or 770 (Right)

Those numbers listed next to the example events are what show up in the marker column. 

Now, I was able to see visual evoked potentials (VEPs) with external triggering (using D17, VDD, GND digital input pins) from a parallel port on a computer with E-Prime software. However, the output of the text file seems to have FIVE columns after the first 8 columns of EEG data. Those 5 columns are linked to each of the digital input pins available, but I only need one of those pins (D17). If LSL does work with at least one digital input, in MATLAB, when I run ReceiveData.m, I would expect to see the 9th column of data containing 0s and 1s for whenever my checkerboard pattern (from the paradigm) switched.

Now, I'm not sure how this might work if I were to use a more complex paradigm. For example, my group designed a motor imagery paradigm (because we're trying to control a computer cursor, in real time, with our headset) in E-Prime, but the events in the paradigm will need their own individual inputs, meaning, we'll need to use more than one of the digital input pins (let's say all 5 for now), if we were able to stream markers over LSL. This is why we switched over to OpenViBE, by the way. Consequently, LSL would need to be able to show that there are 5 columns of markers, AFTER the first 8 columns of EEG data. I'm not sure if LSL will accept all of those columns, and then the timestamp after those. Just a thought.

aj-ptw, 2018-04-02:
I guess we would need to make a special mode for this.

There is a need to move the LSL dependency from the GUI into the HUB, so wher we do that we can output in this specific format. 

Does that sound good to you?

faheemersh, 2018-04-02:
Yes, that's great! Thank you for getting back to me!

closed 2018-04-02 by faheemersh