OpenBCI/OpenBCI_GUI #315: Valid sample data for next GUI (playback demonstration)

Elisabeth-github, 2018-03-12:
## Problem

Provided sample data file of GUI (latest version - December 2017) is void.
At "OpenBCI_GUI/application.linux64/data/EEG_Data" the provided "SDconverted-2016-12-17_18-28-40.txt" is of 0 bytes.

## Expected 

Following William Croft's advisement at this thread http://openbci.com/forum/index.php?p=/discussion/comment/7793#Comment_7793, could I request valid sample data for the next GUI?

If it is possible to provide a (github) link to the sample file for this #BrainAwarenessWeek, that would be great, as we could post on social media that one could download the OpenBCI GUI and load a file for playback to familiarize themselves with an EEG! Before maybe trying to perform an EEG themselves!

Thank you in advance

aj-ptw, 2018-03-13:
Cool! Thanks for the heads up, I did not know this was a need! This is a great first issue for a new contributor to OpenBCI, if no one solves this before the next iteration. @conorrussomanno do you have an ultracortex setup and ready to record?

aj-ptw, 2018-04-02:
Hi we are working on making a repository filled with data to download and experiment with.

Elisabeth-github, 2018-04-02:
@aj-ptw Hi, thank you in advance for your efforts!

PR #515 (2019-05-15): Update sample data and move older data to /deprecated folder
Change gitignore to include sample_data folder

Fixes #315
closed 2019-05-15 by retiutut