OpenBCI/OpenBCI_GUI #316: GUI hangs when trying LSL and when played by Processing IDE, hub does not work

marcelus, 2018-03-16:
## Problem

I'm trying to use my Ganglion with OpenVibe. Now, I'm stuck trying to stream the data to LSL. From the ".exe" GUI, it just hangs when I press to start LSL. So I went to the Processing IDE to get some console message, but when I try to make to use it from the IDE, the GUI shows the message: "Please launch OpenBCI Hub prior to launching this application. Learn at docs.openbci.com". The Hub is already running as administrator.

## Expected 

Connect to OpenVibe.

## Operating System and Version

Windows 10 Home Single Language

## GUI Version

3.2.0

## Running standalone app

Both

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

aj-ptw, 2018-03-22:
Maybe you have two versions of the hub running? Be sure to have no hubs running, then start one hub and then the openbci gui from processing.

marcelus, 2018-03-23:
[image]
[image]

I started OpenBCI_Hub only once and it started two process. Then I've opened the processing and started the GUI from there. Still same issue.
Also, I was running everything in administrator mode.

marcelus, 2018-03-23:
Looking at the code, this is the conditions to the error:

`if (newDataSource != DATASOURCE_SYNTHETIC && newDataSource != DATASOURCE_PLAYBACKFILE && !hub.nodeProcessHandshakeComplete) {`

This line 157 from ControlPanel.pde. I've tried to force "hub.nodeProcessHandshakeComplete" as true when instantiating hub. It goes through the error message and opens the menu, but freezes at line 1201 from the GUI code:

`for (int i = 0; i < hub.deviceList.length; i++) {`

It giver a NullPointerException. Probably cause it's not really connected to the hub.

aj-ptw, 2018-03-23:
There is an issue where the two hubs are open. Force close both and be sure to only start the hub once.

marcelus, 2018-03-23:
I've already do this a couple of times. And same results :/

aj-ptw, 2018-03-23:
Starting the hub starts two processes?

aj-ptw, 2018-03-23:
I have not heard of this problem :(

aj-ptw, 2018-03-23:
Try force closing the hub with the higher process ID

marcelus, 2018-03-23:
Tried to close the process with less memory consumption and start the GUI from processing. Same issue.
And when I try to close the process with more memory consumption it kills the other one automatically.
[image]

sidharthanup, 2018-03-29:
[image]

The OpenBCI_GUI folder is for the standalone app. 
[image]
[image]

sidharthanup, 2018-03-29:
[image]
[image]
[image]
This is when I run the sketch

sidharthanup, 2018-03-29:
Ya the same thing for me as well.

aj-ptw, 2018-04-02:
Looks like the hub might just be running two processes classified by the operating system, so perhaps the two hub processes running is not the problem.

Are any of you running a proxy? Or some software that could prevent the internal communications from working on the mac? 

Did you see a pop-up asking if you would allow private communications for the OpenBCI Hub?

aj-ptw, 2018-04-04:
hey im running hub on my windows machine and it's totally cool that two processes open up when running the hub, this is not the problem

aj-ptw, 2018-04-04:
looks like i have three processes running when the hub is in use by the GUI, are y'all running these hubs as admin?

marcelus, 2018-04-04:
Hey, about the proxy, I'm not running any.
I think that it did poped up a window asking about permitions and stuff, but I granted all access to it.

And yes, I'm running everything as admin :p (HUB, Processing, GUI)

aj-ptw, 2018-04-04:
I can see that the GUI is not able to talk to the hub, so something is blocking this connection. I wonder what it could be. As far as I know both of those check boxes need to be hit. Is this problem duplicating on other Windows computers for you?

marcelus, 2018-04-04:
[image]
seem to be fine. Just opened the Hub and then ran the python script.

aj-ptw, 2018-04-04:
ooouuuu 0.o it's the processing file not being able to talk to the hub!!!

okay! cool we are totally getting somewhere

marcelus, 2018-04-04:
Nice :D

aj-ptw, 2018-04-04:
Okay! same thing, start the hub (if it's not already) and don't run the GUI in standalone or processing and then run [this processing file](https://github.com/OpenBCI/OpenBCI_Hub/blob/master/tests/status.pde).

Expected output:
```processing
q,200,;
q,200,;
```

_note_ idk why it's printing twice but status is good

aj-ptw, 2018-04-04:
thanks for your help btw

sidharthanup, 2018-04-04:
[image]
Status is good

aj-ptw, 2018-04-04:
Okay next up, [uncomment this line](https://github.com/OpenBCI/OpenBCI_GUI/blob/master/OpenBCI_GUI/InterfaceHub.pde#L296) and run the GUI from processing!

sidharthanup, 2018-04-04:
Ran it after uncommenting the line in InterfaceHub.pde 
[image]
Couldn't find any changes

aj-ptw, 2018-04-04:
show me all of the output from the processing windows

sidharthanup, 2018-04-04:
[image]

marcelus, 2018-04-04:
I got the same output as sidharthanup printed.

Then I uncommented line 296 from InterfaceHub.pde and here goes the screenshot:
[image]

marcelus, 2018-04-04:
Ops, I didn't see that there was already an answer.
But yet, same results here.

aj-ptw, 2018-04-04:
Try using an older version of Processing like 3.3.4, maybe 3.3.7 is broken?

sidharthanup, 2018-04-04:
Yes! Its working now
[image]
Never occurred to me to try an older version..Thank you!

marcelus, 2018-04-09:
Sorry for de delayed answer. I was getting a Null Pointer Exception, but I found out that the error was from a piece of code that I had modified.
After getting it back to the original code, it worked out with Processing 3.3.6.

Thanks!! 😄 😄

trtnk, 2018-04-19:
I am having similar problems as well. I want your help.

* Enviroment
    - OS: windows 10
    - Proccessing version: 3.3.4
    - Type of board: Cyton
    - WiFi use: No
    - Proxy use: Yes

In this environment, when I run OpenBCI_GUI from Proccesing and click Live(from Cyton),  
It will look like the image below.

[image]

I show the image of the error message displayed by Processing also.  
OpenBCIHub is running in administrator mode.

[image]

Is Proxy the cause?

trtnk, 2018-04-23:
Thanks for speedy response to the question.

I tried STANDALONE application, I got below message at the application window of OpenBCI_GUI.
```
Please launch OpenBCI Hub prior to launching this application. Learn at docs.openbci.com
```

When I tried running the OPENBCI GUI from Procissing IDE, I got below message at console of Processing IDE. And it came up to the same result as the STANDALONE application.
```
Could not parse -1 for --display
Welcome to the Processing-based OpenBCI GUI!
Last update: 9/5/2016
For more information about how to work with this code base, please visit: http://docs.openbci.com/OpenBCI%20Software/
Graphics & GUI Library: ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
attempting to load images for presentation...
DONE loading images!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
java.net.SocketException: Malformed reply from SOCKS server
    at java.net.SocksSocketImpl.readSocksReply(SocksSocketImpl.java:129)
    at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:459)
    at java.net.Socket.connect(Socket.java:589)
    at java.net.Socket.connect(Socket.java:538)
    at java.net.Socket.<init>(Socket.java:434)
    at java.net.Socket.<init>(Socket.java:211)
    at processing.net.Client.<init>(Unknown Source)
    at OpenBCI_GUI$Hub.startTCPClient(OpenBCI_GUI.java:12568)
    at OpenBCI_GUI$Hub.<init>(OpenBCI_GUI.java:12544)
    at OpenBCI_GUI.systemUpdate(OpenBCI_GUI.java:982)
    at OpenBCI_GUI.draw(OpenBCI_GUI.java:482)
    at processing.core.PApplet.handleDraw(PApplet.java:2439)
    at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:849)
    at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
    at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
    at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
    at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
    at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
    at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
    at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
    at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
    at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
    at java.util.TimerThread.mainLoop(Timer.java:555)
    at java.util.TimerThread.run(Timer.java:505)
Instantiating hub object...
click
click
click
```

What should I try next?

aj-ptw, 2018-04-23:
Are you running the OpenBCI hub?

trtnk, 2018-04-23:
Yes I am. OpenBCIHub is running in administrator mode.  
Three processes run after running OpenBCIHub.

neurochems, 2018-05-10:
hi all, jumping in to say that i was having the same issues with the same conditions as marcelus and sidharthanup and a downgrade of processing from 3.3.7 to 3.3.6 also worked for me. thanks!!

commit 72877da0 (2018-05-16): FIX: #316 processing not working with 3.3.7
closed 2018-05-16 by aj-ptw