OpenBCI/OpenBCI_GUI #317: Networking with LSL makes GUI freezy

rodrigohubner, 2018-03-16:
## Problem

When I select to send packages through LSL, the GUI freezes and the following trace is shown:
...
LSL selected from Protocol Menu
openBCI_GUI: startButton was pressed...starting data transfer...
startRunning...
DEBUG: start data transfer
port not open
Stream update numChan to 8
nPointsPerUpdate 10
dataToSend len: 80
[TimeSeries, obci_eeg1, EEG, 8, 1]
java.lang.RuntimeException: java.lang.NoClassDefFoundError: com/sun/jna/Platform
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:412)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NoClassDefFoundError: com/sun/jna/Platform
	at edu.ucsd.sccn.LSL.<clinit>(LSL.java:976)
	at edu.ucsd.sccn.LSL$StreamInfo.<init>(LSL.java:123)
	at OpenBCI_GUI$Stream.openNetwork(OpenBCI_GUI.java:24826)
	at OpenBCI_GUI$Stream.start(OpenBCI_GUI.java:24296)
	at OpenBCI_GUI$W_networking.startNetwork(OpenBCI_GUI.java:24122)
	at OpenBCI_GUI$W_networking.mouseReleased(OpenBCI_GUI.java:23888)
	at OpenBCI_GUI$WidgetManager.mouseReleased(OpenBCI_GUI.java:25604)
	at OpenBCI_GUI.mouseReleased(OpenBCI_GUI.java:11854)
	at processing.core.PApplet.mouseReleased(PApplet.java:2785)
	at processing.core.PApplet.handleMouseEvent(PApplet.java:2694)
	at processing.core.PApplet.dequeueEvents(PApplet.java:2607)
	at processing.core.PApplet.handleDraw(PApplet.java:2448)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:849)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
Caused by: java.lang.ClassNotFoundException: com.sun.jna.Platform
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:335)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 24 more

## Expected 

To send packages via LSL perfectly :)

## Operating System and Version

Linux Debian 9 64bit

## GUI Version

3.2.0 for Linux 64bit

## Running standalone app

Downloaded app from download site

## Type of OpenBCI Board

Cyton / Cyton+Daisy / SYNTHETIC (algorithmic)

## Are you using a WiFi Shield?

No

aj-ptw, 2018-03-22:
Hi this is a known issue, please run the GUI from processing

compmonks, 2018-04-01:
I have got a similar issue:

java.lang.UnsatisfiedLinkError: Unable to load library 'liblsl64.dylib': Can't obtain...

Operating System and Version
oSX High Sierra 10.13.3

GUI Version
3.2.0 Dec 2017
running from Processing

Type of OpenBCI Board
Cyton Live 16 Channels

Are you using a WiFi Shield?
Nope

aj-ptw, 2018-04-02:
@compmonks any solution to this yet?

compmonks, 2018-04-02:
@aj-ptw unfortunately not. Is this reproducible? I tried to download again the LSL-java library from [labstreaminglayer repo](https://github.com/sccn/labstreaminglayer/tree/master/LSL) but the error persists.

aj-ptw, 2018-04-02:
Are you sure you are downloading the right version for mac? Maybe you are downloading the windows version. 

You can always build the LSL binary from source

aj-ptw, 2018-04-02:
Keep in mind @compmonks the error this person issue is experiencing is in linux, not mac. So perhaps a new issue should be opened for your problems.

retiutut, 2019-03-26:
Also happening on Windows:

retiutut, 2019-04-06:
**The above error happens on Windows and MacOS when running the GUI in STANDALONE**

*Confirmed using ConsoleLog Feature*

retiutut, 2019-04-06:
Fixed!!! 🔥 🔥 🔥  Tested on Mac using GUI. Tested on Windows using GUI and OpenVibe.
[image]

PR #471 (2019-04-14): Development V4.1.2 beta.0
- Enhancement: Update Playback History Widget
- Add Feature: Expert Mode
- Add Feature: Clear GUI Settings Button w/ Confirmation
- Optimize playback history
- General GUI UX/UI improvements
- Fixes #480
- Fixes #482
- Update make-release.py to Python 3 and minor edit to OS X build process
___
- Add missing LSL dependency 💻 🔥 
- Fixes #317 
- Fixes #320 
- Fixes #344 

Tested on Mac, Windows, and Linux.
____

- [x] Update Changelog , info.plist.tmpl, and OpenBCI_GUI.pde to reflect changes
- [x] Request Review
closed 2019-04-19 by daniellasry