OpenBCI/OpenBCI_GUI #318: Using the standalone OpenBCI GUI to stream via LSL freezes the application

sidharthanup, 2018-03-29:
## Problem

The LSL stream freezes in the standalone GUI and when I ran the GUI using Processing and clicked on Live(from Ganglion) it gives an error "Please launch OpenBCI Hub prior to launching this application". Is there a fix to this?

## Operating System and Version

Windows 10

## GUI Version

OpenBCI GUIv3.2.0

## Type of OpenBCI Board

Ganglion

aj-ptw, 2018-03-29:
Yup follow the tutorial:
http://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI#the-openbci-gui-running-the-openbci-gui-from-the-processing-ide

closed 2018-03-29 by aj-ptw

sidharthanup, 2018-03-29:
But I did exactly as it was instructed. I ran theOpenBCI Hub before running the GUI via processing and still the same error.