OpenBCI/OpenBCI_GUI #319: Networking widget LSL, no Ganglion high pass filtering?

CristhianEd, 2018-04-01:
## Problem

If Ganglion has a built-in high-pass filter (in the hardware) at 0.3 Hz (3 dB cut), why do dynamic DC offset appear in the signals from -5000 to 500uV in MATLAB / LSL?

## Expected 

The data received in MATLAB through LSL should not have DC offset.

## Operating System and Version

Windows 10 Pro

## GUI Version

3.2.0

## Running standalone app

Processing 3

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

aj-ptw, 2018-04-02:
I think the data that leaves the GUI from LSL is not using the bandpass filter like is seen in the GUI.

Could you clarify what you mean by there should be no DC offset? Are you saying the DC offset should be removed by hardware?

Thanks!

biomurph, 2018-04-02:
Data that leaves the GUI in any form is directly what it gets from the hardware (Cyton, Ganglion).
There is invariably a DC offset due to the electrode / skin connection. If you want to remove the offset, you can apply a simple high pass filter to the raw data.

aj-ptw, 2018-04-02:
Okay I'm going to close this issue for now because it is not a bug within the GUI.

Thanks @biomurph

closed 2018-04-02 by aj-ptw