OpenBCI/OpenBCI_GUI #320: Networking in LSL error (macOS)

compmonks, 2018-04-02:
## Problem

Following similar issues on other platforms (ex: https://github.com/OpenBCI/OpenBCI_GUI/issues/317), an error freezes the GUI when trying to stream data with LSL. This is happening in my case on mac OS:

`
_LSL selected from Protocol Menu

openBCI_GUI: startButton was pressed...starting data transfer...

startRunning...

DEBUG: start data transfer

Cyton: startDataTransfer(): writing 'b' to the serial port...

Hub: sendCommand(char): sending 'b'

Hub: processCommand: success -- 41664

Stream update numChan to 16

nPointsPerUpdate 5

dataToSend len: 80

[TimeSeries, obci_eeg1, EEG, 16, 1]

java.lang.UnsatisfiedLinkError: Unable to load library 'liblsl64.dylib': Can't obtain InputStream for darwin/liblsl64.dylib
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:408)
	at java.lang.Thread.run(Thread.java:748)
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application._
`

## Expected 

LSL stream output

## Operating System and Version

oSX High Sierra 10.13.3

## GUI Version

3.2.0 Dec 2017

## Running standalone app

running from Processing 3

## Type of OpenBCI Board

Cyton+Daisy 16 chan

## Are you using a WiFi Shield?

No

aj-ptw, 2018-04-04:
Did you put the libraries into your local folder?

daniellasry, 2019-04-19:
fixed

PR #471 (2019-04-14): Development V4.1.2 beta.0
- Enhancement: Update Playback History Widget
- Add Feature: Expert Mode
- Add Feature: Clear GUI Settings Button w/ Confirmation
- Optimize playback history
- General GUI UX/UI improvements
- Fixes #480
- Fixes #482
- Update make-release.py to Python 3 and minor edit to OS X build process
___
- Add missing LSL dependency 💻 🔥 
- Fixes #317 
- Fixes #320 
- Fixes #344 

Tested on Mac, Windows, and Linux.
____

- [x] Update Changelog , info.plist.tmpl, and OpenBCI_GUI.pde to reflect changes
- [x] Request Review
closed 2019-04-19 by daniellasry