OpenBCI/OpenBCI_GUI #323: OpenBCI_GUI can't find ganglion

kallander, 2018-04-06:
## Problem
I'm trying to connect the ganglion to the GUI.. The stand alone program works fine. 
I've installed the 3.3.6 version of Processing and everything is where it should in the folders (i think, according to the guide here https://github.com/OpenBCI/OpenBCI_Processing ). The GUI works if I use a synthetic source instead so it's just the connection to the ganglion that's missing.. 

## Operating System and Version

Windows 10 

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

kallander, 2018-04-06:
[image]

aj-ptw, 2018-04-06:
That's the version of the GUI before ganglion was support. Please follow [this OpenBCI GUI tutorial](http://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI). I'm closing this issue.

aj-ptw, 2018-04-06:
Use the code in this repository

closed 2018-04-06 by aj-ptw