OpenBCI/OpenBCI_GUI #328: FEATURE REQUEST: Save Configuration File

emolyzr, 2018-05-01:
## Problem

If I have a fixed experimental setup, I need to set the settings every time again, when I restart the GUI.

## Expected 

I'd expect to do change some settings in the GUI, e.g. notch filter, amplitude range, which channels are (de-)activated and so on. Now, I have the possibility to "save" this settings to a text file. This text file can be set as default (so it is automatically loaded upon restarting the GUI next time) or it can be selected and loaded within the GUI.

## Operating System and Version

macOS/Windows/Linux

## GUI Version

Dezember 2017

## Running standalone app

Downloaded App.

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

emolyzr, 2018-05-20:
Thanks for your engagement! 
As a first step it might be sufficient to save custom settings in a text file that can be read in when restarting the GUI (automatically or via the menu). Probably then user friednliess can be the next step ;)

emolyzr, 2018-05-20:
I know the software is open source, so I am not expecting anything. 

But user friendliness *for me* would be if this function would exist at all in whichever way - because as you said: functionality is a necessity - and if necessary functions would exist at all, I am as a user would be very delighted. But I believe the next 1000 recordings, I have to continue doing roughly 30 clicks each time, before I can start a recording. I hope, I don't make too many mistaktes...

Sorry for the tone... I hope you get the point... Have a nice sunday.

retiutut, 2018-05-20:
I already have in mind a few ways to do this, but it gets complicated quickly. Adding this feature little by little is likely the best approach.

It might be appropriate to run the main function in WidgetManager? That way we only save configurations for widgets that are in use [isActive].

User Preferred:
widgetmanager.currentContainerLayout //default layout 
dataprocessing.currentNotch_ind //default notch
w_analogread.startingVertScaleIndex //default vert scale for analog read widget
w_timeseries.startingVertScaleIndex //default vert scale for time series widget

Activating/Deactivating channels are found in BoardCyton ln 447-481 and also HardwareSettingsController.

changeChannelState()
deactivateChannel()
activateChannel()

Also, have a look here:
https://github.com/OpenBCI/OpenBCI_GUI/blob/16129254c346c5f21d930f4d29657a70c7c9d686/OpenBCI_GUI/Interactivity.pde#L101
There are shortcuts already built in!

For deactivating channels 1-16:
1,2,3,4,5,6,7,8
q,w,e,r,t,y,u,i

For activating channels 1-16, simply hold the Shift key and press the same keys as above:
!, @, #, $, %, ^, &, *
Q,W,E,R,T,Y,U,I

emolyzr, 2018-05-20:
Thank you very much @retiutut for the hint with the shortcuts to deactivate channels!! That will surely save some seconds and nerves ;)
Unfortunately, I am an end-user and doesn't have the time nor skill to do the hacking myself. Thus, I really appreciate any help like this.

retiutut, 2018-05-20:
Please restrict your comments to only those that are helpful in nature. Criticizing others way of thinking and coding is not appropriate. An "end-user" should not be dictating the creative process of others.

emolyzr, 2018-05-20:
> thread [2].
> Links:

Hi,

I am sorry that we caught us on the wrong foot. I agree with you that 
functionality is a necessity and user freindliness is priority. And I 
was serious thanking you for the hint with the channel deactivation. I 
*really* didn't knew that before. I also don't want to be a dictator - I 
am sorry, if it's that what you understood from my message. I really 
appreciate that you try to put this forward. So we are on the same side, 
I guess.

I just don't like that you critizise my "way of thinking" on false 
grounds. If the people who coded that GUI would think of user 
friendliness as a priority they'd have implemented this option right at 
the conception phase. So I found it kind of sarcastic that you say user 
friendliness has priority. I also do not understand why you give thumbs 
down for my comment, it was a helpful comment in that it is really 
necessary and user friendly to have this function in the most simple 
way. I hope you understand that I as an "end-user" can only give 
creative and helpful comments by giving feedback what is helpful at the 
site of the "end-user".

So, I guess everything is said. I continue doing my clicking and - if 
you like - you can continue your creative processes. Let me know, if 
there's something I can help you with (*serious offer*).

Best,
Johannes

---
--
- - -

Dipl.-Psych. Johannes Meixner
Geschäftsführer | Technischer Leiter

http://emolyzr.de

emolyzr GmbH
Institut für Psychologie
Humboldt-Universität zu Berlin
Rudower Chaussee 18, 12489 Berlin
Raum 2'104, 2.OG, Glasturm

T: +49 30 12084543
M: +49 170 5360027

Amtsgericht Charlottenburg HRB 154824, USt-IdNr. DE292836742
Geschäftsführer: Johannes Meixner & Dr. André Weinreich

retiutut, 2018-05-20:
Please be more respectful of the hard work of others. Your comments continue to be attacking and passive-aggressive.

retiutut, 2018-05-20:
@emolyzr or others:
Please post any additional features that should be included in the Save Configuration file that will be made and I will try to find the variables that needed to make it work. This way we can start adding the feature with settings that most Users are using.

emolyzr, 2018-05-20:
In particular *deactivating channels* and changing *hardware settings (especially BIAS, SRB 2, and SRB 1)* would be helpful (as these are important for the recordings). Thank you @retiutut

retiutut, 2018-05-22:
I expect this feature will be ready sooner rather than later!
[image]

retiutut, 2018-05-22:
https://github.com/retiutut/OpenBCI_GUI/blob/development/OpenBCI_GUI/SoftwareSettings.pde

retiutut, 2018-05-22:
https://github.com/retiutut/OpenBCI_GUI/blob/22a3db1b745acfb4aa47363e0dc1c83db5377ad2/OpenBCI_GUI/SoftwareSettings.pde#L81

retiutut, 2018-05-25:
I've completed the first version of this feature! @aj-ptw 

[image]
[image]

aj-ptw, 2018-05-28:
Awesome! Will do a code review this Tuesday!

retiutut, 2019-01-05:
@daniellasry This has been fully implemented as of v4.0.0. Settings for any new widgets will need to be added to SoftwareSettings.pde for saving and loading, using the framework in place.

daniellasry, 2019-01-05:
Nice! I think we should make a note [in the docs](http://docs.openbci.com/Tutorials/16-Custom_Widgets) about adding new settings to SoftwareSettings.pde so I opened the issue above for tracking.

closed 2019-01-05 by daniellasry