OpenBCI/OpenBCI_GUI #329: Problem with OpenBCI_GUI 3.3.0

wellton, 2018-05-07:
## Problem

do not working in Debian GNU/Linux

## GUI Version

The version is 3.3.0

See in the picture.

thanx

att
[image]

aj-ptw, 2018-05-07:
i'm making a linux computer as we speak to work on this.

wellton, 2018-05-07:
show. Thanks =)

chelobaka, 2018-05-09:
Same error on Debian. Version 3.2 worked fine.

wjcroft, 2018-05-09:
AJ, thanks for looking into the 3.3.0 problems.

They are also showing up on Windows as well, in the form of freezing.

http://openbci.com/forum/index.php?p=/discussion/1594/openbci-3-3-0-doesn-t-work-windows-10-64-bit

http://openbci.com/forum/index.php?p=/discussion/comment/8315/#Comment_8315

chrisjz, 2018-05-11:
I've submitted a fix that stops the freezing in Windows 10, which I think should theoretically also work in Linux too.

Until this change gets verified and accepted, you can grab it from here:
https://github.com/chrisjz/OpenBCI_GUI/archive/development.zip

You will need to run it using Processing though, here are the steps (using the above ZIP file instead of the one mentioned in instructions):
http://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI#the-openbci-gui-running-the-openbci-gui-from-the-processing-ide

aj-ptw, 2018-05-12:
Just released a new beta with the bug fix from @chrisjz https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.3.1-beta1

cc: @wjcroft

closed 2018-05-12 by aj-ptw

chelobaka, 2018-05-12:
Original report was about Debian. Version 3.3.1-beta1 doesn't fix it. Now there's an error right after launch:

java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at processing.net.Client.run(Unknown Source)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at OpenBCI_GUI.clientEvent(OpenBCI_GUI.java:12546)
	... 6 more

Clicking on "LIVE (from Cyton)" doesn't show config menu. After each click there's a "click" message in terminal.

oshBTC, 2018-05-16:
had same problem today (freezed when clicked anything) on win10. downloaded the beta and it fixed it. thanks

pspressman, 2018-06-04:
I'm still having a related problem on Windows 10, with both 64 and 32 bit versions.  The latest update, GUI v 3.3.a-beta1, does not seem to help...  still get the same javascript error referenced here: https://github.com/OpenBCI/OpenBCI_Hub/issues/50