OpenBCI/OpenBCI_GUI #331: OpenBCI_GUI v3.3.0 freezes at first click

TComp, 2018-05-13:
## Problem

OpenBCI_GUI starts normally and then freezes at the first click. "Data Source" menu options get highlighted correctly when hovering over with the mouse but the programs freezes no matter where I click, on a menu option or somewhere else on the program window. OpenBCI_Hub seems to work correctly. Both apps are running as administrator and the issue happens with or without firewall. 

## Expected 

I am using the previous version 3.2.0 without any problem.

## Operating System and Version

Windows 10 Home 64 bit version 1803 build 17134.48

## GUI Version

3.3.0 of April 2018

## Running standalone app

Downloaded 64 bit version app

## Type of OpenBCI Board

Cyton with firmware v.3

## Are you using a WiFi Shield?

No

TComp, 2018-05-13:
Just looked at the closed issues and tested version 3.3.1-beta 1, which does not freeze and works fine. The OpenBCI Hub included in the zip file with the 3.3.1 beta 1 version still throws a Javascript extension "cannot find module "openbci-ganglion", but I downloaded it from the website (same 1.4.2 version) and it worked correctly. Thank you.

wellton, 2018-05-14:
The @TComp is right The 3.3.1-beta-1 works =) show...

By the way, the design error appeared. (See the pictures below, please)
[image]

[image]

oshBTC, 2018-05-16:
gui seems to work with the beta. however, I get error messages when I try to start the hub from this version. using the old hub now.

aj-ptw, 2018-05-17:
@oshBTC @TComp 

I just fixed the download for 3.3.1, you can download that, it has the working hub.

Sorry!

andrewjaykeller, 2018-07-16:
Closing for now as I think v3.3.2 solves this, please let me know otherwise!

closed 2018-07-16 by andrewjaykeller