OpenBCI/OpenBCI_GUI #334: BIAS no longer included

emolyzr, 2018-05-21:
## Problem

Started streaming with BIAS "included". Sometimes after a while all channels will show 0,10µV, but are still running and reacting. A look to the hardware settings reveals that BIAS has changed to "Don't include". Don't know how this happened. Already encountered this problem multiple times. May I accidentally pressed a short cut do exlcude BIAS?

## Expected 

This should better not happen.

## Operating System and Version

WIN 10 64bit PRO v1709

## GUI Version

December2017, v3.2.0

## Running standalone app

app

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

retiutut, 2018-05-21:
Using the current keyboard shortcuts, disabling a channel in the Time Series widget will automatically turn Bias to "Don't Include" and SRB2 to "Off". Similarly, turn the channel back back on _this time using the shift key + keyboard shortcut_ and Bias will change to "Include" and SRB2 to "On". However, it seems that pressing the channel button on and off _does not affect_ Bias and SRB2 options.

retiutut, 2018-05-21:
There are some old lines of code in BoardCyton that mention biasAuto and biasFixed, but it says they are "now depreciated". These shortcuts don't appear in Interactivity and seem to have been removed for some good reason. Also, not sure what they used to do, just that it had to do with Bias.

https://github.com/OpenBCI/OpenBCI_GUI/blob/16129254c346c5f21d930f4d29657a70c7c9d686/OpenBCI_GUI/BoardCyton.pde#L52

https://github.com/OpenBCI/OpenBCI_GUI/blob/16129254c346c5f21d930f4d29657a70c7c9d686/OpenBCI_GUI/BoardCyton.pde#L53

emolyzr, 2018-05-21:
First: To use the shortcuts correctly, you need to use the ENG keyboard Layout. My GER layout produced strange outcomes.
Second: Using the shortcuts for activation and deactivation consistently will change BIAS and SRB2 ... *as long as you didn't make any manual changes in these hardware settings*. As soon as you set BIAS to "Don't include" yourself, using shortcuts won't affect BIAS and SRB2. BUT... acutally it seems that these shortcuts are not reliably changing SRB2 and BIAS, e.g. if I press "1" for deactivating channel 1, then BIAS and SRB2 turn to off. I "accidentally" press "1", channel 1 ist still deactivated and SRB2 and BIAS didn't change. I reactivate channel 1 by pressing SHIFT + 1, but SRB2 and BIAS don't change.

However, this doesn't seem to speak to my problem. Because channel 1 is activated the whole recording and only BIAS seems to switch from "Include" to "Don't include"... so it's not accidentally deactivating a channel by using shortcuts. But as you said there aren't dedicated shortcuts for changing the BIAS, right?

retiutut, 2018-05-21:
The only keyboard shortcuts that appear to be active are those in OpenBCI_GUI/OpenBCI_GUI/Interactivity.pde . I am experiencing the same weird control issues, and I'm using v3.3.1 development branch. @emolyzr maybe try updating to v3.3 April 2018 or newer.

Would adding a shortcut to make "all BIAS to Include" across all channels help? I can possibly try to add a shortcut.
I've identified the lines of code in HardwareSettingsController starting here:
https://github.com/OpenBCI/OpenBCI_GUI/blob/16129254c346c5f21d930f4d29657a70c7c9d686/OpenBCI_GUI/HardwareSettingsController.pde#L127

retiutut, 2018-05-21:
I've created a shortcut to turn Bias On all channels using the lowercase 'L' key! 

Let me see if I can fix this channel keyboard shortcut mixup.

retiutut, 2018-05-21:
[image]
[image]

retiutut, 2018-05-22:
[image]

[image]

retiutut, 2018-05-22:
[image]
[image]

retiutut, 2018-05-22:
@emolyzr I see the problem with the channel shortcuts with the German Keyboard vs English Keyboard! The issue is with the top row of numbers when Shift key is pressed. I was able to learn this here http://ascii-table.com/keyboard.php/129

retiutut, 2019-01-05:
@daniellasry This issue was also resolved quite some time ago.

Heads up, there may be a few more issues I can recommend for final review and closure.

daniellasry, 2019-01-05:
@retiutut thanks! yes, please do recommend any issues you think should be closed or looked at.

closed 2019-01-05 by daniellasry