OpenBCI/OpenBCI_GUI #338: Accelerometer display pops up error then only gives 0 value readings

x37v, 2018-06-06:
## Problem

The accelerometer display pops up error and displays 0 for all 3 axes.

Reproduction steps:

1) start the gui
2) connect to board and start
    * we used the computer's internal BLE
3) select the accelerometer display tab
4) click "turn accel on"

You should now see an error popup window: 

`OpenBCIHub Fatal Error`
`"list" must be an Array of Bufers`

[image]

5) click "ok" on the error popup
6) click "Start Data Stream"

We see only 0 for the 3 accelerometer values.

## Expected 

We would expect no error popup and valid accelerometer readings.

## Operating System and Version

macOS 10.12.6

## GUI Version

3.3.0

## Running standalone app

standalone

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

aj-ptw, 2018-06-18:
Working on a patch for this bug!

x37v, 2018-06-18:
Fantastic! Thanks @aj-ptw

commit 92eb94ea (2018-06-20): FIXES: #338
commit 96306b6c (2018-06-20): Merge pull request #342 from aj-ptw/development

FIXES: #338
closed 2018-06-20 by aj-ptw