OpenBCI/OpenBCI_GUI #339: Cyton Board Data Stream Problem

lars3107, 2018-06-12:
## Problem

I use the Cyton Board and whenever I start the Data Stream, multiple channels, especially channel 1, show hight amplitudes. Additionally when I turn off channel 1, the Data Stream stays the same.

[image]

## Expected 

Since I followed the tutorial "Cyton Getting Started Guide" I wouldn't expect these issues. A few weeks ago I didn't have these issues.

## Operating System and Version

Windows

## GUI Version

3.2.0

## Running standalone app

I tried both methods and I am having the same issue with both.

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

lars3107, 2018-06-13:
I found the solution for my problem. After I updated the firmware of the Rfduino on the dongle(HOST) and on the Cyton board(DEVICE), everything worked as it should.

retiutut, 2019-04-07:
@daniellasry This issue is can be closed.

closed 2019-04-07 by daniellasry