OpenBCI/OpenBCI_GUI #340: GUI Freezes 193 seconds into SD recording - recordings end early

GeneralFox2, 2018-06-13:
## Problem

I noticed that the GUI freezes after a few seconds when recording to the SD card.  The unit also seems to stop recording as well.  In this example, it only recorded 193 seconds before the GUI stopped showing the readings.  But the interface remained operational, and I was able to stop the recording, and close the session.  I have attached the recorded file.
[OBCI_13.TXT](https://github.com/OpenBCI/OpenBCI_GUI/files/2096252/OBCI_13.TXT)

If you look at the file in a raw file viewer, you'll see it does not finish recording a full sampling:
- On D5 it only records 6 electrodes prior to closing the data
- You'll notice that a large segment of the file remains available for data to be saved in it after the data is closed - why was the file not more filled?

I attached the converted file.  Please convert the file to have a .csv extension to return it to it's original file format.
[SDconverted-2018-06-12_19-54-06.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/2096271/SDconverted-2018-06-12_19-54-06.txt)
If you review it, you'll note that it also records the samples of the 6 electrodes on the last data line (213) and then closes the data.  Also, in the recorded file there are 12 numbers at the end - but in the SD Converted file there are only 7 - this is non-consequential as this data is not properly formatted anyway, so it should be ignored by any analysis or players.

## Expected 

In a perfect world, if I set the interface to record for a maximum of 5 minutes, it would record for 5 minutes (which would fill the file space allocated), finish the recording on the SD card, and close the file.  It would not end the recording prematurely.  Coffee and donuts would be free and I would drive a Ferrari.

## Operating System and Version

Windows 10

## GUI Version

3.3.1 

## Running standalone app

Downloaded OpenBCI GUI

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

No

GeneralFox2, 2018-06-13:
Another instance.  In this example, I ran out of time in the recording so the interface stopped showing the samples.  I closed the session.  Unfortunately, I cannot extract the data using the Converter in the GUI as it just freezes.  If you review the attached file you'll see the file is filled with data, but the data was not closed off at the end of the file.

[OBCI_17.zip](https://github.com/OpenBCI/OpenBCI_GUI/files/2096435/OBCI_17.zip)

aj-ptw, 2018-06-14:
I know there is code on the cyton that stops the recording once the SD time has finished, perhaps there is a miscalculation in the amount of time the SD file is recording for?

@biomurph do you have any ideas?

GeneralFox2, 2018-06-15:
I was wondering how that worked.

On a slightly parallel topic
I am hoping that this means the Cryton can work independently of the computer?  I was hoping that I could:
- configure the Cryton using the GUI
- start the recording
- then walk away from the computer, out of the range of the dongle, and go to the nearby park.

I wanted to move away from any localized interference (I live by a full sized unlit soccer field) during the recording session.  If the Cryton does not allow this, then can you modify the code to enable this functionality?  Communicating with the computer when recording to SD should not be necessary.

retiutut, 2019-04-19:
@GeneralFox2 A lot has changed since 3.3.1. Any old SD-card-files may need to be converted again. We have improved the playback files significantly, so some older files may not work, but SD card files haven't change much.

Try using https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.1.2-beta.0! 

> I am hoping that this means the Cryton can work independently of the computer? I was hoping that I could:
>
> configure the Cryton using the GUI
> start the recording
> then walk away from the computer, [...]

Can you try this again and let me know what happens?

closed 2019-05-12 by retiutut