OpenBCI/OpenBCI_GUI #345: cannot detect cyton over bluetooth on mac

mgroth0, 2018-06-28:
## Problem

I cannot detect the cyton board using OpenBCI_GUI on mac. Over Windows, the board can be detected just fine.

## Expected 

I expect to see the Cyton board on the SERIAL/COM PORT list under "Live (from Cyton)". Right now I'm not seeing it whether or not the bluetooth USB dongle is in (could somebody also clear whether or not I should be using the dongle?). All things on the list are other devices except /dev/cu.Bluetooth-Incoming-Port and /dev/tty.Bluetooth-Incoming-Port.

I did install FTDI 2.4.2

## Operating System and Version

macOS 10.13.3

## GUI Version

Version 1

## Running standalone app

Standalone

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

retiutut, 2018-06-29:
The Cyton requires the Dongle to connect over bluetooth. 
Follow the instructions here: http://docs.openbci.com/Tutorials/01-Cyton_Getting%20Started_Guide

The dongle address will look something like the one shown in this picture.
https://github.com/OpenBCI/Docs/blob/master/assets/images/serial_cyton_select_serial_port.png?raw=true

retiutut, 2018-06-30:
New Video Tutorial describing the same process:
https://youtu.be/D5RCu41msT4

mgroth0, 2018-07-02:
It turns out my version is actually 3.1.1. But when you click "About OpenBCI GUI" from the menu bar on a mac, the version number is displayed as 1.

Also, I think I am doing the process right. When I do the same process on windows with the same board and dongle, I can connect just fine. But on my mac, the port you pictured above does not appear.

I repeated the process with the same board and dongle on another mac. On that mac the port pictured above _did_ appear, but froze on "Start System".

retiutut, 2018-07-02:
Ok. 

*The "About OpenBCI GUI" on Mac (maybe other OS) needs to be updated for next release* cc: @aj-ptw 

@mgroth0 I'm glad you had other machines to test with! _Sometimes after I put my iMac to sleep, I can't see the USB serial port I was using for the dongle._ I learned this was a hardware/OS issue of some kind because the port may or may not be missing when I go to look for all serial ports with Max or 3rd party serial reader. The easy fix for my issue is to either plug the dongle into a different USB port or restart my iMac. 
- Current Mac OSX version is 10.13.5. Try update both Macs.
- On the Mac that froze on Start, does it do this every time? three times in a row?

mgroth0, 2018-07-02:
I updated my computer to 10.13.5, but I have the same problem. I don't think sleeping my computer is an issue because I never saw the USB serial port in the first place, and haven't slept my computer once since it restarted after updating. I've also tried both ports.

On the other computer where it was freezing, we tried restarting the GUI. Both times it froze. We also never slept that computer after plugging in the dongle.

retiutut, 2018-07-02:
_It definitely sounds like you are experiencing an issue with both of these Macs and the GUI._ @aj-ptw should be able to help more than I can at this point.

mgroth0, 2018-07-03:
I tried installing and running the Processing version of OpenBCI GUI.

I am having an issue with this version before I can even see the list of connected devices. When I click on "LIVE (from Cyton)", the window to the right that is supposed to pop up does not appear. I used to have this same issue on the PC, until I figured out it was because I wasn't running the OpenBCI Hub beforehand. But now on my mac, I have moved the openBCIHub.app I just downloaded from [here](https://github.com/OpenBCI/OpenBCI_Hub/releases) (version 1.4.4) into my applications folder, and I always run it before trying to start OpenBCI GUI from Processing (even though the [tutorial]("http://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI#the-openbci-gui-running-the-openbci-gui-from-the-processing-ide") says "If this is your first time running the sketch, you will get a message from the Mac OS that will ask you if you will allow permission for an unsigned app to run. The app in question is the Electron Hub"). So presumably the app tries to start the Hub on its own, even if I don't. But it appears either way that there is a problem connecting to the hub. 

Here's the full log when I start OpenBCI GUI from Processing.
```

2018-07-03 16:15:35.969 java[30656:4302388] pid(30656)/euid(501) is calling TIS/TSM in non-main thread environment, ERROR : This is NOT allowed. Please call TIS/TSM in main thread!!!
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Mac
Welcome to the Processing-based OpenBCI GUI!
Last update: 9/5/2016
For more information about how to work with this code base, please visit: http://docs.openbci.com/OpenBCI%20Software/
Graphics & GUI Library: ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
attempting to load images for presentation...
DONE loading images!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
java.net.ConnectException: Connection refused (Connection refused)
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at java.net.Socket.connect(Socket.java:538)
	at java.net.Socket.<init>(Socket.java:434)
	at java.net.Socket.<init>(Socket.java:211)
	at processing.net.Client.<init>(Unknown Source)
	at OpenBCI_GUI$Hub.startTCPClient(OpenBCI_GUI.java:12830)
	at OpenBCI_GUI$Hub.<init>(OpenBCI_GUI.java:12806)
	at OpenBCI_GUI.systemUpdate(OpenBCI_GUI.java:1002)
	at OpenBCI_GUI.draw(OpenBCI_GUI.java:485)
	at processing.core.PApplet.handleDraw(PApplet.java:2426)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:866)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
Instantiating hub object...
-- UDP session started at Tue Jul 03 16:15:39 EDT 2018 --
-- bound socket to host:127.0.0.1, port: 51000 --
[18-07-03 16:21:13.188 -0400] close socket < port:51000, address:127.0.0.1 >

SHUTDOWN HOOK
SHUTDOWN HUB
```

retiutut, 2018-07-03:
Yes, the app does try to start the Hub. @mgroth0 To have the GUI launch the hub with Processing, move OpenBCIHubb.app to OpenBCI_GUI/data/

mgroth0, 2018-07-10:
I moved OpenBCIHub.app into OpenBCI_GUI/data but the board still doesn't show in the list of serial/COM ports. 

```
2018-07-10 15:45:17.188 java[2834:133926] pid(2834)/euid(501) is calling TIS/TSM in non-main thread environment, ERROR : This is NOT allowed. Please call TIS/TSM in main thread!!!
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Mac
Welcome to the Processing-based OpenBCI GUI!
Last update: 9/5/2016
For more information about how to work with this code base, please visit: http://docs.openbci.com/OpenBCI%20Software/
Graphics & GUI Library: ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
attempting to load images for presentation...
DONE loading images!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
Instantiating hub object...
Hub: clientEvent: handshake complete
click
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
setInterface: curInterface: 0
Setting sample rate for Cyton to 250Hz

sample rate set to: 250
Transfer Protocol set to serial
```

andrewjaykeller, 2018-07-17:
Did you install the FTDI drivers?

mgroth0, 2018-09-06:
We've been using the Cyton on Windows with no connection issues (and Ganglion on mac), so this issue is not a priority for me anymore. But I tried all the steps anyway in case I could provide useful information. Unfortunately nothing worked.

1) reinstalled FTDI 2.3 from [here](https://www.ftdichip.com/Drivers/VCP.htm) multiple times

2) tried both USB ports

3) reboot multiple times

4) used coolterm and after scanning always shows the same thing as OpenBCI GUI:

[image]
[image]

5) verified dongle blue LED light is on

6) verified switch is on GPIO6

daniellasry, 2019-04-19:
@mgroth0 I know this is no longer a priority, but I would love to know if this is still happening the GUI v4.1.2 beta. We're focusing on eliminating these first-time connection issues.

closed 2019-06-10 by retiutut