OpenBCI/OpenBCI_GUI #347: No Data Stream - Cyton V3, Serial, GUI 3.3.1, Win 10

zecahue, 2018-06-30:
## Problem

I am unable to get any data after the Data Stream is started.
No data from time series nor from the accelerometer windows.
I am using a Cyton V3 (serial-dongle) GUI 3.3.1, HUB started as administrator, windows 10.
Serial/COM port- Status "Success: System is Up"
Battery 3.7V 1200 mAh (checked, shows 4V in the meter)

All previously firmware procedures was successful.

All channels 0.00 uVrms
Accelerometer 0g

Even if I touch at the female pins and turn the board to other directions, all remains zero.

## Expected 

I expect some data streaming.

## Operating System and Version

Windows 10

## GUI Version

3.3.1

## Running standalone app

I am running the downloaded app.

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

retiutut, 2018-07-01:
Follow directions [here under Troubleshooting](http://docs.openbci.com/Tutorials/01-Cyton_Getting%20Started_Guide). When the GUI initializes properly a success message shows at the bottom of the GUI: "The GUI is done intializing. Cyton firmware v3.x.x Press \"Start Data Stream\" to start streaming!" It's possible you are losing the connection prior to starting the stream or the Cyton is off.  Data will stream when the Cyton is connected properly.

zecahue, 2018-07-01:
Hi, thanks for your reply.

Yes, I get the ""The GUI is done intializing. Cyton firmware v3.3.1 Press "Start Data Stream" to start streaming!" So, my guess is that the connection is ok. Battery is fully charged. Is there a way to test if the ads1299 is ok? 

I'm trying the openbci forum as well to detected the problem.

retiutut, 2018-07-01:
__The GUI does not show that message by accident, so I can conclude that the GUI is working properly.__ I can't see how the GUI could be the problem. Please seek help in the forum if you followed all of the troubleshooting steps above.

zecahue, 2018-07-01:
Yes I did all the steps, so I will try something else. I will close this issue, thanks.

closed 2018-07-01 by zecahue