OpenBCI/OpenBCI_GUI #348: Array Out of Bound for playback file

ahsanic, 2018-07-02:
## Problem

When I select the playback file, provided by OPENBCI. I get array out of bound. And GUI crashes.

Debug Logs:
%OpenBCI Raw EEG Data % %Sample Rate = 250.0 Hz %First Column = SampleIndex %Other Columns = EEG data in microvolts
OpenBCI_GUI: initSystem: loading complete.  125130 rows of data, which is 2147483647 seconds of EEG data
OpenBCI_GUI: initSystem: Initializing core data objects
EEG_Processing: *** ERROR *** Filters can only work at 125Hz, 200Hz, 250 Hz, 1000Hz or 1600Hz
EEG_Processing: *** ERROR *** Filters can only work at 125Hz, 200Hz, 250 Hz, 1000Hz or 1600Hz
EEG_Processing: *** ERROR *** Filters can only work at 125Hz, 200Hz, 250 Hz, 1000Hz or 1600Hz
EEG_Processing: *** ERROR *** Filters can only work at 125Hz, 200Hz, 250 Hz, 1000Hz or 1600Hz
EEG_Processing: *** ERROR *** Filters can only work at 125Hz, 200Hz, 250 Hz, 1000Hz or 1600Hz
EEG_Processing: *** ERROR *** Filters can only work at 125Hz, 200Hz, 250 Hz, 1000Hz or 1600Hz
OpenBCI_GUI: initSystem: -- Init 1 -- 26265
OpenBCI_GUI: initSystem: Initializing FFT data objects

retiutut, 2018-07-02:
Please post an issue in the format provided. There is not enough information. Please post the beginning of the playback file and the first few lines of data.

wjcroft, 2018-07-02:
http://openbci.com/forum/index.php?p=/discussion/1434/array-out-of-bounds-eeg-processing-error

ahsanic, 2018-07-31:
No solution yet?

retiutut, 2018-07-31:
Please update to GUI version 3.4.0 and upload the playback file.

ahsanic, 2018-08-01:
I have also tried GUI Version 3.4.0 same error. 

I have attached the screenshot too.
[image]

HamzaHummam, 2018-08-01:
I am having the same issue in version 3.3.4 while running it from processing in windows 10 ..... ?

retiutut, 2018-08-01:
Again, please upload playback files being used. You will get an error when using sample EEG playback files because they are old. There is already an issue to fix this. https://github.com/OpenBCI/OpenBCI_GUI/issues/315

retiutut, 2018-08-01:
This issue can be closed using the information from a duplicate thread. https://github.com/OpenBCI/OpenBCI_GUI/issues/368

You are using the sample playback data that is old.

> You will get this same error when using sample EEG playback files because they are old. There is already an issue to fix this. #315

HamzaHummam, 2018-08-02:
[openBCI_2013-12-24_meditation.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/2252380/openBCI_2013-12-24_meditation.txt)  I have used this file.

ahsanic, 2018-08-02:
Can you please send a the new data file?

retiutut, 2018-08-02:
New data files will be available when #315 is resolved. It is already marked as In Progress.

**This issue can be closed**

retiutut, 2019-01-05:
@daniellasry This issue can be closed.