OpenBCI/OpenBCI_GUI #349: GUI Rendering delay causes artifact pulses on Widgets

MobileRez, 2018-07-02:
## Problem

While using the Open BCI Gui with the Time Series, Band Power, & Head Plot widgets I would get random artifact data in the device across all channels and all configurations. This is also duplicatable when just using the head plot widget.

Noticed there is no system setting to set recourse priority with the APP, this is causing it to not be able to access all the resources it needs, not sure if the fact its build with java has any impact with that, though would explain the unreasonable high demand on resources. 

## Expected 

No artifact nose is GUI from rendering issues, able to control System Resource priority from software as later versions of Win 10 do not let you set it from task manager therefor you cannot access all system resources as needed.

## Operating System and Version

Windows 10 Pro x64
Version: 1709
Build: 16299.431

## GUI Version

3.3.1

## Running standalone app

Running in OpenBIC GUI 

## Type of OpenBCI Board

Cyton running firmware v3.1.1

## Are you using a WiFi Shield?

No

andrewjaykeller, 2018-07-17:
Okay so this is all because of the headplot :(

Did you try changing the smoothing at all? I think the headplot needs to be rewritten!

MobileRez, 2018-07-22:
Yes and no setting I could use on that Head Plot graph could help smooth things out. I know most people would call my surface book (1) with performance base old by now but its got some decent power too it and a decent graphics card as well so it should be able to handle this rendering... that must be some really bad code...

paquettejp, 2019-04-10:
Hi,

Wondering if there has been any development on this issue? It seems to be similar, if not the same issue as: https://github.com/OpenBCI/OpenBCI_GUI/issues/231

I am using a Cyton with WiFi Shield (firmware, 2.0.5) and I have also noticed these intermittent bursts of noise in my recordings. I am running the OpenBCI standalone GUI (Version 3, 4.0.3) on OS X 10.13.6.

To check the FPS issue, I used the Time Series and Head Plot separately with the Widget Template and Marker Mode. The Time Series seemed to stay in the high 50s but the Head Plot got down to 10 FPS.

In both cases, I observed this burst noise.

**Time Series:**
[image]

**Head Plot:**
[image]

retiutut, 2019-04-10:
It’s packet loss related to the WiFi shield and Cyton, hardware related. Issue is described in detail already. #231

retiutut, 2019-04-11:
@paquettejp the issue here, about Headplot FPS, is definitely on the table to be fixed soon!

paquettejp, 2019-04-11:
> @paquettejp You are experiencing packet loss related to the WiFi shield and Cyton, hardware related. Issue is described in detail already. #231 The team is definitely aware of this and we are still working on a plan of action, up to designing new hardware.
> 
> This issue here is likely caused by Headplot widget being greedy, so not related to packet loss.

Thank you for clearing that up! Two separate issues.

closed 2019-06-10 by retiutut