OpenBCI/OpenBCI_GUI #350: cannot connect to Cyton on Windows (reconfiguring)

mgroth0, 2018-07-04:
Just to be clear, I have created a separate issue for the problems I have connecting the board to a mac. The problem I have with windows is different, so I'm creating a new issue.

## Problem

After "START SYSTEM", the GUI hangs for a bit while its setting up. Then after about 5 to 10 seconds it just stops trying to connect and goes back to the way the program looks when it first starts up and you are choosing a DATA SOURCE. 

## Expected 

I expect to see graphs and eeg waves, but none of that happens for even a moment. I might also expect an error popup or something, but that doesn't happen either.

## Operating System and Version

Windows 10 Home version 1709 

## GUI Version

The version is displayed on startup.

## Running standalone app

Processing

3.3.2

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Processing Log

The log says something about "Halting system for reconfiguration of settings", but it does this every time.

```

Welcome to the Processing-based OpenBCI GUI!
Last update: 9/5/2016
For more information about how to work with this code base, please visit: http://docs.openbci.com/OpenBCI%20Software/
Graphics & GUI Library: ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
attempting to load images for presentation...
DONE loading images!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
Instantiating hub object...
Hub: clientEvent: handshake complete
click
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
setInterface: curInterface: 0
Setting sample rate for Cyton to 250Hz

sample rate set to: 250
Transfer Protocol set to serial
click
ControlPanel — port is open: false
initSystem yoo

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: initSystem: -- Init 0 -- 13402
timeOfInit = 13402
OpenBCI_GUI: initSystem: Preparing data variables...
OpenBCI_GUI: initSystem: Initializing core data objects
OpenBCI_GUI: initSystem: -- Init 1 -- 13404
OpenBCI_GUI: initSystem: Initializing FFT data objects
OpenBCI_GUI: initSystem: -- Init 2 -- 13409
OpenBCI_GUI: initSystem: Closing ControlPanel...
OpenBCI_GUI: initSystem: Initializing comms with hub....
OpenBCI_GUI: hub : Sent connect to Hub - Id: COM6
OpenBCI_GUI: initSystem: -- Init 3 -- 14410
  3a -- 14410
  3b -- 14410
cyton: openNewLogFile: opened ODF output file: SavedData\OpenBCI-RAW-2018-07-03_19-23-30.txt
ChannelController: createChannelSettingButtons: creating channel setting buttons...
accel w 1022
accel h 630
accel x 1
accel y 109
comPorts = [COM4, COM5, COM6]
init analog read bar 0
init analog read bar 1
init analog read bar 2
widgets.size() = 13
widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
wm.widgetOptions.size() = 13
widgetOptions:
[Time Series, FFT Plot, Accelerometer, Networking, Band Power, Head Plot, EMG, Focus Widget, Pulse Sensor, Digital Read, Analog Read, Marker Mode, Widget Template 1]
Powering 3 widgets up, and remapping.
Activating widget [0]
Activating widget [1]
Activating widget [2]
accel w 510
accel h 292
accel x 512
accel y 446
  3c -- 15537
OpenBCI_GUI: initSystem: -- Init 4 -- 15537
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
OpenBCI_GUI: stopRunning: stop running...
Closing any open SD file. Writing 'j' to OpenBCI.
disconnecting serial
-- UDP session started at Tue Jul 03 19:23:31 GMT-05:00 2018 --
-- bound socket to host:127.0.0.1, port: 51000 --
[18-07-03 19:31:44.459 -0500] close socket < port:51000, address:127.0.0.1 >

Client SocketException: Socket closed
```

retiutut, 2018-07-04:
The number after a checkpoint -- Init # -- is a running counter in milliseconds.There should be success commands printed to the console from the hub between initialization checkpoints 2 and 3.  I don't see them! Have a look at what happens on my Mac with the Processing: 

`OpenBCI_GUI: initSystem: -- Init 2 -- 26542
OpenBCI_GUI: initSystem: Closing ControlPanel...
OpenBCI_GUI: initSystem: Initializing comms with hub....
OpenBCI_GUI: hub : Sent connect to Hub - Id: /dev/tty.usbserial-DM00D7TW
Hub: processConnect: made it -- 27212 code: 200
Hub: setBoardType(): sending 'cyton -- 27212
Hub: processBoardType: success -- 27213
OpenBCI_GUI: initSystem: -- Init 3 -- 27543`

retiutut, 2019-05-04:
@mgroth0 Let us know if this still happens in the latest release of the GUI.

closed 2019-05-04 by retiutut