OpenBCI/OpenBCI_GUI #351: Not converting MicroSD Card recording

GeneralFox2, 2018-07-05:
## Problem

Was recording when noticed Cyton not recording properly.  EEG waveforms on the OpenBCI screen looked all the same and looked like the letter "S".  The unit was not railing.  I stopped the recording using the interface.  I started a second recording and ended it after a while.  

I removed the SD card and tried to use interface to convert the first recording from the SD Card - but when I do the GUI freezes.  If I click on the X on the GUI to close the screen then I notice a small 2,784 KB file is created.  File should be much larger.  I used the interface to convert the second recording - it worked as expected and was very fast.

## Expected 

The GUI would convert the first recording to a usable format.  The file would be as large as the recording.  I have attached the first recording to this post.

[OBCI_1D.zip](https://github.com/OpenBCI/OpenBCI_GUI/files/2164680/OBCI_1D.zip)

## Operating System and Version

Windows 10.

## GUI Version

3.3.1

## Running standalone app

Stand-alone App.

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

No

andrewjaykeller, 2018-07-17:
Thanks for providing the zip file, running now.

andrewjaykeller, 2018-07-17:
Okay so this file just froze my computer GUI!

andrewjaykeller, 2018-07-17:
I found the issue, there is invalid data in the end of the file which throws and error on `DataLogging.pde Line 1565.`

GeneralFox2, 2018-07-17:
What causes the invalid data to be written? 

⁣Thank you,
Don​

andrewjaykeller, 2018-07-17:
Stopping the recording early. I think the firmware allocates the space on the file before over writing the data again.

GeneralFox2, 2018-07-19:
> Okay, so that weird signal is a test signal, it's a square wave.
> Somehow it's being sent to the board.

Why would it suddenly do that? 

Is there something wrong with my brain?   (Be nice)

⁣Thank you,
Don​

GeneralFox2, 2018-07-19:
Thinking about this,  this test signal starts from time to time during recordings.   It happens during good recordings,  no railing.   Why? 

⁣Thank you,
Don​

commit 7eeb1934 (2018-07-17): Merge pull request #360 from andrewjaykeller/development

FIX: #351
commit bbcd532a (2018-07-17): FIX: #351