OpenBCI/OpenBCI_GUI #356: Networking widget: data stream freeze through OSC

akselmb, 2018-07-10:
## Problem

I'm sending data from the networking widget through OSC to Max 7. The data streams work perfectly for around one minute, but then stops/freezes. When I switch between the programs/windows, from Max 7 to OpenBCI_GUI and back, the stream continue to flow. Not really sure if this is OpenBCI_GUI problem or a Max7 problem.

## Expected 

The stream to flow continuous without any freezes.

## Operating System and Version

macOS 10.11.6

## GUI Version

v.3.3.2

## Running standalone app

Running the downloaded app

## Type of OpenBCI Board

Ganglion, but doesn't really matter. Have tried playback and synthesized as well.

## Are you using a WiFi Shield?

No

retiutut, 2018-07-19:
@akselmb I use the networking widget to stream 16 channels of data via OSC to Max 7 using the latest GUI. The latest Mac OS is 10.13.5. Also, a fourth OSC stream will be available soon with GUI v3.4.0. This is because OSC is easy to work with and reliable. Make sure Max is up to date with the latest version of OpenSoundControl object/package. This is likely a Max problem, check if errors are printing to console in Max. Errors print at the bottom of the GUI or in the console with the Processing version. 

Here is a simple patch to accept data from the GUI:

`<pre><code>
----------begin_max5_patcher----------
2111.3oc6akzbaibE9rzuBT7PNQqz6K4VFev0jpbxT1yjToljREDXKNvFKr.
Ark8Ty7aO8B.HAEDH.HnLnxnCTfM5l8266s1OB9qWe0h6RePscg2ew6m8t5p
e85qtxNjYfqJe+UKh8eHHxeqcZKBRiiUI4KV5tWt5gb63uOOS4G6QptQRQbZ
QdjJ2tLP4nay+RjxN+p4ceZR91vuZGDxtAr2m6GJ1lGdeXfedXZh89k2bied
vuDlr91LUPtC7LL8F5ROBR+I3I3lWQja.d+2cvIL4QnoXylL01s16r+FDtxh
wz69vqj6KoAoQoYtMDbijwEH7RyUbjjSrWIkBIeoGzr0l08aWes4kkSC6h+F
wtDF77vth4D6h9FwtHg77vt74D6B+FwtPD1vnSO6xdVY2XMZ7WqZmDgOII5l
T9W1nbfawhZAusnn.hkeb+CBcrE5oXKziIE5h4prQvmprQlsxFhJNQYCOakM
oM0yIHZnwHZIpOqW6ihk8Od+qeUlF+Ju+b5FUxcAgsK7nAH7K6mSITWgSeXf
1pgQLkTfVteeZQxpWqiOmkF4gA5+ZmFvOhFVot2uHJua1Xokn+wvX0O5utWb
S4+nfAyMPvTZeTrRm1HPE9IkIuCQ3EjD6m2WGjiRN8fJX1HAPW5tAxD7KbOk
xP7mfmBjwtv4fxTAmBG.DW3bfKkADCGKEPYuTCXVRM.vXiWhwmuvkzmwvkNh
X7QKwjWplHUQPFeNUzYzFg8LZiTxDi2HA8h0HoJU6IT30YrtK9ynQRISbB0c
MJukmpaKFqju60e+su4m9du7Tu25+f2eWk+4zrOpgs2OXve0ZhBSTAZCJ6BI
Cr6L24G7w0YFyw8EqF8rQN5d1vkLSzYrzwoVpUBmhl1.qaP8cqazyFLiKXLS
mZHTHVJMWw.bAkT1ylmpYOBFhSnloK3.pP3tRPvvouUZuKL3W7yV48u7ixi8
S9OI+shnunOaLTzpNEM85TvopSqJL0oTQxIQoBuDTpa7STQKlDURKzK05q37
X3XaTHHhNni9TumoYqTYMIRHmQ3V5iRj.jq2l5gfMIx3zUp8+78SV6DHDemk
y5L+Uvg9IaVDpZQ.cQ8Lq5AKM5SyUbJRRNXQaxR2jlUYbZl8d5M6rrNLG7Ef
YYWy3MUlaSKxBpRIT1VE8FWuYqTZmgjZWgetpgnZH8D1K8cizGG43aDZB1Hc
QsGeizm.vNoSamvN31M2wlfcB0GYBMExDpOxj4z0m9VY5SwwkJxDXR.w8Xif
noPlP8g9XSgMQsRnaC8of9.79vefIg.A8gAkO19yEIzeylOox1VNa6tnSf8A
WR.9R6aCSbu0d9hEYpOEVMepcD+LcdobcRohLWdrGDtzJ1LCYIEg1rCWajuq
Ky2cPz2D+X2R+qEqBSeetedw1aeqJovkdptX8FbwcquOLJpNk0U02QmaurN9
Et6trb3Fo2LYNf1rHXHlirULf0WPo6wmkqoNyEoNeEotZAh.go1u4OvAKCsa
u.PoaG.RAf3tROTybWsm6rsTZ0yuHO0jkLT4p9CXG2phczs0TJSe2CKyXQrl
PCqVy9ldMUFwgObpJgZ.1NmhKy7quxTrkkbHGvKijNOPgOwJgojqMmYCh7dm
ZcQjeoE63+ztqHOOM46hJTcq6FV8dShf5f1OTjsIZPfS6XRr0xCqq9iXpXmN
kfqDI25VpaUkcenA37CBzaPC.xnZiJK4Y9yQikgHZ8TG2FlXh7ppKuc4dur+
JZPDvx6t6klRtNZPos6NvphTwGhVyQj3ty5nOHovcEgBPnladiE0OHpkaLiH
bLfVEQsFRLg1yV1Bhc1CMQ7AOXEZ0s0YUXnSafBHkIX3tAg.AvtsV6r6.AWC
KHaDFLvdYv7J3eXx7GlLGKFSr+1bU1K+PL8DsiybYHZJ0WW4G76GGvGs+OmD
hOBarOf2FEpqa97vvDFDBbksJQLhsbHpDJLYTGsIwZ+vje+bfWniZELHgXWI
jIQsUNxGOeb1wDd0W8WcQYd0V3yIJfeo9RRAtftSnW78EIAUEe+Dg8OGjkMd
daFWiXmOlkT0iVWi8YLQyO4ij0SU+gmKqeNWm6ikMjyF6mmm45Owow3CywYe
DTDaZywTel5+uQAN5J2Kc1dSlRkTW39D2hiw0uhZkHjQ4DWiQzYpotDfjxPl
bxKqlMUpOVmFsZVpLLMBATVtDBRs9Fvxk8BTaDlDF6G0klnkCnfq5QClTEHo
puM6GtrshM5Xo5RUIsWczt4RMsv0FkiVuJtndUa0aos5gmnhLsaFjaURvxN.
qG6DNPUc26OntikG7pbPIs6nY16BbevEcXQWU7YiqzElhlnlu1OCZ4QMnaOI
6QQ+z3Grq7OS2T+S+aUTT5m6xs3oZDg4mGjk0.Bh.B6r.ht5Dq1vTNQBmqsC
F4p6Dfs4o+rzq38.HZtCP7bGfj4N.oyY.lk94jA6krCW6xJgOqHDM6QHd1iP
xLFgu9K9C2LDfjTWm93bLFwcMFGBg3oGf837TsgPK6gM+45xmozryE3PyYvg
myfiLmAGcdBt2pOTaRt+fcIDD.i4dpRX.M3rtuLITRmdL5ppc3PDQgb2IsfB
W+do5HMGzJ88eThM+x7MSB.DmCA.coK.3KcAfboK.zKcAfcIJ.Ew2oOf8vKa
3XOR3SH5dyLrnFGvRiVMXr0U2LOK.DM2AHdtCPxbGfzYL.e2v8PpeVTg0cYF
fv5nKSI5r+9f9mgolemSmuG04td3IA2fFyC6L7k1C6blqKuu4He+SMO4OQWh
tr73zbt6f05bmxteNyddL7MeqEpUg4GShN8ut6N+lwXipHA8Z9sq+eTrVB3B
-----------end_max5_patcher-----------
</code></pre>
`

akselmb, 2018-07-25:
Thanks for the reply, @retiutut ! I haven't tried the CNMAT-OSC, so will try this now :) Any suggestions to where I should connect [print] in Max in order to troubleshoot?

retiutut, 2018-07-25:
"OSC is a content format developed at CNMAT by Adrian Freed and Matt Wright comparable to XML, WDDX, or JSON." - [https://en.wikipedia.org/wiki/Open_Sound_Control](url)

Errors print to Console in Max and Processing.

retiutut, 2019-03-26:
@daniellasry This issue can be closed.

closed 2019-03-30 by daniellasry