OpenBCI/OpenBCI_GUI #361: BUG: Cannot see the numbers,

andrewjaykeller, 2018-07-17:
## Problem

I want to be able to see the numbers, but they are black background on the individual channel numbers.

## Expected 

to be able to read the channel numbers on the time series widget. 

## Operating System and Version

macOS/Windows/Linux

## GUI Version

Latest

## Running standalone app

Both

## Type of OpenBCI Board

Cyton/Cyton+Daisy/Ganglion

## Are you using a WiFi Shield?

Yes

MobileRez, 2018-07-17:
https://developer.paciellogroup.com/resources/contrastanalyser/ is a good tool to check colors and accessibility to make sure the largest sample size of people can read the new color choices (and the tool is free!)

retiutut, 2018-07-17:
This bug addressed in Development 3.4.0 Alpha 3! https://github.com/OpenBCI/OpenBCI_GUI/pull/362
[image]
[image]

retiutut, 2018-07-23:
The current solution is the best option for readability based on my knowledge of special education literacy practices. To increase accessibility, use built-in operating system tools. Try inverting the whole screen colors or using magnification and/or a different screen resolution. There is already an issue regarding High DPI screens: https://github.com/OpenBCI/OpenBCI_GUI/issues/145.

MobileRez, 2018-07-23:
I deal with neurodiversity and most standards for accessibility only address physical needs (screen readers, blindness, low vision, etc). I am also aware of that open issue, but the current windows 10 scaling fix to solve the channel readability, compresses the actual info and makes it hard to read.

retiutut, 2018-07-23:
It's still a separate issue from what was mentioned here. This minor bug has already been resolved. You are welcome to share results from the contrast analyzer and submit helpful suggestions that would improve the results from the analysis. Try using a monitor resolution of 1920x1080 until https://github.com/OpenBCI/OpenBCI_GUI/issues/145 is resolved.

MobileRez, 2018-07-23:
ill just keep working with Andrew on the mobile app test I have started with the BIC boards and the WiFi shield, was more asking questions, but it seems ill have to open more tickets and its not worth the extra work when I'm starting off on the losing end.

retiutut, 2018-07-23:
I'm not sure what that has to do with this thread? Please post helpful comments and suggestions.

MobileRez, 2018-07-23:
***Context for why I am even bothering with posting here***
I was on the call with Andrew when he opened this, which is why i added my notes above. what i was suggesting/asking about with the selected channels in deactivate state and the ohm button(i think that's its name) (gray background, black text in that sate) since it was not shown in the screenshots above.  an answer i was looking for was "that was not part of this fix" or "it was addressed just no screenshot shown" the second comment by the way, would solve any High dpi concerns I have while the fix happens, and will help still when that High DPI fix does go out.   the comment just before this one was made in frustration, and seeing that i was getting the feeling that you did not want to spend any more time on this issue, i was attempting to drop it, but since that was not valid...

Back to my original questions which seems to have gotten lost. did you just update only those numbers in their **active** state, or did this fix the issue of readability when the channels and other buttons are in their **deactive** state? if not, do I need to open more issues against that, or since they will be related issues to this., can they be addressed in this fix and no need to open related tickets and add the extra administrative tasks?

MobileRez, 2018-07-23:
thank you, if you are able to change that, it will help a lot! If not I understand, at least you looked into it.

The version of the GUI i currently have: (3.3.1) all button text (like the Omega button, channel buttons, etc.) or otherwise is black when deactivated, which is why I ask if you fixed this, great, if not, then let me know and I will. if I am misunderstanding what you are saying i.e. the channel button should not be black, or should be fixed in this next update) let me know.

retiutut, 2018-07-23:
Yes, this stuff is very fixable! On the other hand, scaling the app for larger resolutions takes more time. I hope these changes can increase accessibility for everyone and we can extend the best parts of the GUI, like Time Series, to mobile with similar styles. Do you think the channel button size in 3.3.1 is large enough?

MobileRez, 2018-07-23:
sorry about the confusion earlier. I'm a little scared porting over some of these features since I am trying to convert this all to .net and everything currently coded for open BCI is in languages i don't know or cant easily port. so... small steps. https://github.com/MobileRez/MobileRezOpenBCI is the repo where the app code will show up once I get it checked in.

retiutut, 2018-07-23:
[image]

retiutut, 2019-01-05:
@daniellasry This issue has been resolved. See above.

closed 2019-01-05 by daniellasry