OpenBCI/OpenBCI_GUI #371: java/bin/java: not found

brylie, 2018-08-18:
## Problem

After downloading the OpenBCI GUI 3.3.2 for Linux from the website, I attempt to run the application from the command line with 
- `cd path/to/extracted/files/application.linux64`
- `sudo ./OpenBCI_GUI`

At that point, I see the following error:

> ./OpenBCI_GUI: /home/< username >/BCI/OpenBCI/application.linux64/java/bin/java: not found

When I look inside the `java` folder, I see a sub-folder named `java`. Inside the `java` sub-folder is a `bin` folder with the `java` command. So, the package seems to have one level of `java` folders too many.

## Expected 

The application should start without Java error.

## Operating System and Version

Ubuntu 18.04

## GUI Version

3.3.2

## Running standalone app

Downloaded app

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

brylie, 2018-08-18:
Note, moving the contents of the `java` sub-folder to the `java` parent folder fixes the issue.

winsteadworks, 2018-10-16:
@retiutut this is still an issue in the latest edition. Also moving the contents solved the issue of it not loading, however the gui is blank and white when I run it.

retiutut, 2018-10-19:
@winsteadworks Going to try and test

retiutut, 2018-10-19:
> Note, moving the contents of the `java` sub-folder to the `java` parent folder fixes the issue.

**Tested and confirmed this issue and fix using Ubuntu 18.04.1 and OpenBCI_GUI v3.4.0-beta on a MacBook Air**

> however the gui is blank and white when I run it.

I did not experience this issue. Make sure that the  appropriate version of the OpenBCI Hub is set up correctly and run the GUI with `sudo`

retiutut, 2018-10-19:
[image]

I also noticed characters not drawing correctly in the title bar.

retiutut, 2018-10-23:
**It seems this issue has been resolved in the latest release 3.4.0!**

https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/3.4.0
@winsteadworks 
@brylie

winsteadworks, 2018-10-24:
@retiutut well I got the GUI and Hub running, however my issue now is that my GUI does not recognize that the hub is running and is telling me it cannot initialize until I have started the hub

andrewjaykeller, 2018-11-15:
Did you update your hub with GUI?

cosmobille, 2018-11-15:
I don't know what the hub is. I edited the bash script.

daniellasry, 2019-04-19:
Believe this is fixed

closed 2019-04-19 by daniellasry