OpenBCI/OpenBCI_GUI #374: BUG: 3.4.0 not working with WiFi Shield

andrewjaykeller, 2018-08-27:
## Problem

Running the GUI as standalone i don't get an error and the GUI just freezes. 

Here is the relevant output code

```
Ch 0, Active, x24, Normal, Include, On, Off
OpenBCI_GUI: activating channel 1
**
Powering up channel 1
Command: !
Hub: sendCommand(char): sending '!'
Ch 1, Active, x24, Normal, Include, On, Off
OpenBCI_GUI: activating channel 2
**
Powering up channel 2
Command: @
Hub: sendCommand(char): sending '@'
Ch 2, Active, x24, Normal, Include, On, Off
OpenBCI_GUI: activating channel 3
**
Powering up channel 3
Command: #
Hub: sendCommand(char): sending '#'
Ch 3, Active, x24, Normal, Include, On, Off
OpenBCI_GUI: activating channel 4
**
Powering up channel 4
Command: $
Hub: sendCommand(char): sending '$'
Ch 4, Active, x24, Normal, Include, On, Off
OpenBCI_GUI: activating channel 5
**
Powering up channel 5
Command: %
Hub: sendCommand(char): sending '%'
Ch 5, Active, x24, Normal, Include, On, Off
OpenBCI_GUI: activating channel 6
**
Powering up channel 6
Command: ^
Hub: sendCommand(char): sending '^'
Ch 6, Active, x24, Normal, Include, On, Off
OpenBCI_GUI: activating channel 7
**
Powering up channel 7
Command: &
Hub: sendCommand(char): sending '&'
Ch 7, Active, x24, Normal, Include, On, Off
OpenBCI_GUI: activating channel 8
**
Powering up channel 8
Command: *
Hub: sendCommand(char): sending '*'
done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;
done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;
done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;

k,406,Error: socket hang up,;
Hub: processCommand: ERROR_COMMAND_NOT_ABLE_TO_BE_SENT -- 54153 Error: socket hang up
done writing channel.r,set,0,0,24,normal,1,1,0,;

done writing channel.r,set,0,0,24,normal,1,1,0,;
done writing channel.r,set,0,0,24,normal,1,1,0,;

java.lang.RuntimeException: Waited 5000ms for: <39c1ea42, 3706030a>[count 2, qsz 0, owner ] - 
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:410)
	at java.lang.Thread.run(Thread.java:748)
```

## Expected 
The GUI to run and me to be able to stream data from the WiFi Shield

## Operating System and Version

macOS

## GUI Version

3.4.0

## Running standalone app

Both

## Type of OpenBCI Board

Cyton
## Are you using a WiFi Shield?

Yes

retiutut, 2018-08-27:
An attempt was made to catch this error when using Cyton+Dongle. The error catch appears here: https://github.com/OpenBCI/OpenBCI_GUI/blob/2a98e2fb0dff550977d67ace0b2a22f85dd5b9a0/OpenBCI_GUI/SoftwareSettings.pde#L1033
The allotted time window is currently 3800 milliseconds. 

- [ ] Possible fix: maintain time window, add new error catch when socket hang up occurs -- `k,406,Error: socket hang up,;
Hub: processCommand: ERROR_COMMAND_NOT_ABLE_TO_BE_SENT -- 54153 Error: socket hang up`

vikramdhoot, 2018-09-08:
Im also facing the same issue ......

Im getting the data on serial port and im able to detect the device ID also of ads1299 ... but when i connect the wifi shield im not getting the device id also not streaming the data