OpenBCI/OpenBCI_GUI #376: UDP: duplicate packets

mgroth0, 2018-09-06:
## Problem

Recieving duplicate packets when streaming data over UDP network and listening from python.

## Expected 

I expect to receive each sample once.

## Operating System and Version

macOS 10.13.5

The version is displayed on startup.

3.3.2

Are you running the downloaded app or are you running from Processing 3

Running from processing

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

I noticed duplicate samples in my python script which listens to a single ganglion EEG timeseries stream over UDP. I looked for some example code to make sure I configured the socket correctly, and found [udp_receive.py](https://github.com/OpenBCI/OpenBCI_GUI/blob/96306b6c905692e8199e6e3da5b5e76822b75845/Networking-Test-Kit/UDP/udp_receive.py). I tried running this with no modifications, and observed the same problem.

Take a look at what was printed to console, which I copy and pasted into [udp_receive_results.log](https://github.com/OpenBCI/OpenBCI_GUI/files/2358464/udp_receive_results.log). Now take a look at the corresponding [OpenBCI-RAW-2018-09-06_14-28-17.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/2358465/OpenBCI-RAW-2018-09-06_14-28-17.txt). If you search for `1320.7` in the .txt file, you can find the point where I started listening to UDP. 

Perhaps a workaround, for now, could be to create a small buffer in the python code, maybe 20 samples long, and with each new sample received check that the buffer does not contain it. I have not tested this yet, but it does not seem ideal for a couple reasons:
1) What if there actually happened to be the same reading twice very close to each other
2) what if a duplicate is more than 20 samples away? I'll have to run tests and find the maximum distance a duplication will occur.

Is it possible I'm supposed to send back a confirmation to server each time I receive a packet?

retiutut, 2019-05-04:
@mgroth0 Is this still happening with the Latest Release? https://github.com/OpenBCI/OpenBCI_GUI/releases

mgroth0, 2019-05-05:
Can you remind me where the data file is saved? I checked /Applications/OpenBCI_GUI.app/Contents/Java/data/EEG_Data and it was not there

retiutut, 2019-05-05:
Its in Applications/SavedData for now.

black, 2023-02-21:
Can we reopen this? I am getting duplicated over UDP connection in GUI vs 5.1.0 on windows. I have created  an app to receive the data in nodejs using dgram.

retiutut, 2023-02-24:
This would need to be replicated using the networking test kit provided in this repository.

black, 2023-02-25:
@retiutut I was streaming Band Power data over UDP. I tested the streaming when filter was on and off during the streaming.

retiutut, 2023-03-01:
While working on an internal feature, I noticed that the timestamps generated by a modified version of udp_receive.py from the Networking Test Kit are duplicated 100+ times down to the millisecond.

retiutut, 2023-03-15:
We have fully replicated this issue and are now actively developing a fix.

black, 2023-03-15:
@retiutut awesome

retiutut, 2023-03-16:
A fix has been implemented internally and is now in testing for a special project for OpenBCI.

Explanation of issue:
- Networking "Stream" is run in a separate thread
- Stream thread asks for data faster than it becomes available, resulting in 100+ data outputs while the main GUI thread processes one visual drawing frame.

Explanation of fix:
- Add an AtomicBoolean as a "dataIsReady" flag at the end of the loop for the main GUI thread
- When this flag is true, new data is likely available and has been computed from background threads and widgets

retiutut, 2023-04-12:
Fixing this issue for TimeSeries is a little more complicated, but I've managed to get it right while sending out only a single channel (for simplicity). Here I compared the last 10 data points for the channel with the most recent 10 data points, and if they are different send out the new sample. This isn't technically correct, but the chances of 10 float being the exact same is nearly impossible.

retiutut, 2023-04-14:
I've fixed the TimeSeries output and confirmed the desired sampling rate over UDP for all channels. This fix will be available immediately for internal use and ported to the Public GUI after TED.

closed 2023-06-26 by retiutut