OpenBCI/OpenBCI_GUI #378: Unable to easily maximise GUI window in Windows OS on laptop

tristanlbailey, 2018-09-15:
## Problem

Whenever I start the OpenBCI GUI, the program's window doesn't quite fit on my laptop's screen. I believe that my laptop uses a fairly common resolution amongst laptops (1366 x 768), and since the resolution is not overly large, I need to be able to use the entire screen to make good use of the program.
So, I need to be able to resize the program's window. Unfortunately, when the program starts, the top bar of the window goes off of the screen, and so I can't access the Maximise button which usually resides at the top right. While you can manually resize the window, it takes a bit longer to do than using the Maximise button, and the window size and position is not saved. If you close and then open the program again later, the program returns to the previous fixed size (approximately 1040 x 749).
The Alt + Spacebar key trick doesn't work either, as it does with other Windows applications.
The only way I know how to get the Maximise button to appear, is to unlock the taskbar at the bottom of the screen (default position of the taskbar in Windows), drag the taskbar to the top of the screen and release, and then drag the taskbar back down to the bottom of the screen, every time you use the program.
This behaviour is less than user-friendly.

## Expected 

I should be able to see the top / title bar of the OpenBCI GUI, and access the Minimise, Maximise, and Close buttons. Also, the custom size and position of the window should be saved when set by the user, so that the program reverts to the same size and position when it is run next time.

## Operating System and Version

Windows 8.1

## GUI Version

All versions, including the latest version (3.4.0)

## Running standalone app

I am running the downloaded app.

## Type of OpenBCI Board

Cyton (not that it matters)

## Are you using a WiFi Shield?

No

andrewjaykeller, 2018-09-15:
@retiutut could this be a saved option?

retiutut, 2018-09-15:
@andrewjaykeller The default render option during setup in https://github.com/OpenBCI/OpenBCI_GUI/blob/master/OpenBCI_GUI/OpenBCI_GUI.pde#L365 is **1024x768 with 2D rendering**. This means the vertical resolution is already at screen max for @tristanlbailey's laptop. This should work: 

`  println("Screen Resolution: " displayWidth + " X " + displayHeight);
  //Set the GUI size based on screen size, can be expanded later to accomodate high res/dpi screens
  //If 1366x768, set GUI to 976x549 to fix #378 regarding some laptop resolutions
  if (displayWidth == 1366 && displayHeight == 768) {
    size(976, 549, P2D);
  } else {
    //default 1024x768 resolution with 2D graphics
    size(1024, 768, P2D);
  }`

retiutut, 2019-03-18:
@daniellasry This issue can be closed. Confirmed working on a Lenovo Thinkpad w/ touchscreen.

closed 2019-03-19 by daniellasry