OpenBCI/OpenBCI_GUI #386: Unable to connect to Cyton (MacOS/Processing 3.4/OpenBCI GUI 3.4.0)

coreygo, 2018-10-07:
## Problem

My OpenBCI previously worked without issue on MacOS and Windows with OpenBCI up until v3.4.0. The OpenBCI now seems to have an initialization timeout when attempting to connect.

Problem thread on the OpenBCI forum here:
https://openbci.com/forum/index.php?p=/discussion/comment/9338

Processing console logs here:
- v3.4.0: https://gist.github.com/coreygo/949f49c7daf2689351514a33adb22d63
- v4.0.0-alpha.1: https://gist.github.com/coreygo/a06cea595e8bd4a8f1f0300682bc4367

## Expected 

Initialization and data acquisition from Cyton.

## Operating System and Version

macOS 10.13.6

## GUI Version

- v3.4.0 (stand alone and via Processing) w/Hub v1.4.5
- v4.0.0-alpha.1 w/Hub v2.0.0 via Processing

## Running standalone app

Both stand alone and via Processing 3.4.0

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No, USB serial w/ftdi-vcp-driver v2.4.2

retiutut, 2018-10-09:
Thank you for posting the full log files @coreygo!

Tagging @andrewjaykeller

coreygo, 2018-10-10:
No problem at all. Thanks for investigating!

I've found the time to do some additional testing and am still stumped a bit.

Here's a new file containing both go/no go states  (sorry it's truncated, Processing's console history isn't long enough):
- v4.0.0-alpha.1 in Processing: https://gist.github.com/coreygo/acfbe452819966708e9639d313d25bcb

- I was able to resolve the initialization timeout by disabling writing to the SD card.
- If I change the setting to write any maximum length of time the initialization fails.

Another, likely separate issue, is with detection of the USB serial port. I'm noticing that I often have to restart my MBP or switch to another USB port in order for Processing/OpenBCI to detect insertion of the dongle and refresh the port list.

Running either `ioreg -p IOUSB -w0 -l` or "System Information" with the dongle in any/all of my USB ports does show the device as present but OpenBCI doesn't necessarily refresh the port list to display the dongle unless I switch to a previously unused port or reboot the system.

Getting everything in sync seems to be somewhat finicky but when it does work… things are great :relaxed:

Here are the steps I take to reproduce a go state:

1. Boot system.
2. Open Processing and load OpenBCI GUI.
3. Click Allow to run OpenBCI Hub.
4. Insert USB serial dongle with switch set to GPIO_6.
5. Power Cyton with switch set to PC.
6. Refresh port list.
7. Click Status. If system displays:
```
Refreshing port list...
Radios_Config: system_status
Radios_Config: connect_to_portName: attempting to open serial port: /dev/tty.usbserial-DN0096YR
Radios_Config: connect_to_portName: port is open!
S1299 Device ID: 0x3E
LIS3DH Device ID: 0x33
```
Then power cycle the Cyton board and repeat. System should now display:
```
Radios_Config: system_status
OpenBCI V3 16 channel
On Board ADS1299 Device ID: 0x3E
LIS3DH Device ID: 0x33
```
8. Click Do not write to SD.
9. Start System.
10. Go for capturing data.

Hope that helps. Please let me know if I can be of any assistance.

retiutut, 2018-10-10:
> Another, likely separate issue, is with detection of the USB serial port. I'm noticing that I often have to restart my MBP or switch to another USB port in order for Processing/OpenBCI to detect insertion of the dongle and refresh the port list.

Running either ioreg -p IOUSB -w0 -l or "System Information" with the dongle in any/all of my USB ports does show the device as present but OpenBCI doesn't necessarily refresh the port list to display the dongle unless I switch to a previously unused port or reboot the system.

coreygo, 2018-10-10:
I should also mention that I'm on macOS 10.14 (Mojave) now but was able to confirm the go state prior to upgrading from 10.13.6. I also ran `brew cask reinstall ftdi-vcp-driver` and rebooted post upgrade to be sure everything is okay to go. The latest log file is after all that jazz.

retiutut, 2018-10-10:
> I should also mention that I'm on macOS 10.14 (Mojave) now but was able to confirm the go state prior to upgrading from 10.13.6. I also ran `brew cask reinstall ftdi-vcp-driver` and rebooted post upgrade to be sure everything is okay to go. The latest log file is after all that jazz.

**So you can confirm that the GUI can connect via Dongle with 10.14 (Mojave)?**

coreygo, 2018-10-10:
Yep, as long as I do not write to the SD card.

Writing to the SD card still reproduces the no go initialization timeout as shown in the log here: https://gist.github.com/coreygo/acfbe452819966708e9639d313d25bcb

retiutut, 2018-10-10:
I just reproduced what you've mentioned here.

retiutut, 2018-10-10:
**v4.0.0-alpha.2
When using SD card mode of any duration, the initialization timeout window passes and the system halts.**

Here is a log file to document this bug:
https://gist.github.com/retiutut/a703325744e548beb153356038321322

retiutut, 2019-03-04:
[image]

closed 2019-03-13 by daniellasry