OpenBCI/OpenBCI_GUI #389: Connection of Cyton Board to OpenBCI GUI

hfuhr, 2018-10-19:
## Problem

The Cyton board isn't connecting correctly with the OpenBCI GUI and the GUI stream will delay for several seconds at a time. The accelerometer data will come but the EEG data doesn't appear to work. The EEG data shows 8 identical perfect curves and doesn't react to blinking or frowning. The GUI used to show changes when this happened. I am also using the OpenBCI EEG Headband for the 8 electrodes. Does anyone know how to resolve this problem? I have already tried resetting the board multiple times and checking the connections. 

## Expected 

In a perfect world I hope to be able to collect accurate data with the OpenBCI GUI. Any help is greatly appreciated!

## Operating System and Version

macOS High Sierra Version: 10.13.4

## GUI Version

GUI v3.3.2

## Running standalone app

Running the downloaded app

## Type of OpenBCI Board

Cyton Board

## Are you using a WiFi Shield?

No

hfuhr, 2018-10-26:
[OpenBCI-RAW-2018-10-26_15-31-20.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/2520835/OpenBCI-RAW-2018-10-26_15-31-20.txt)
[OpenBCI-RAW-2018-10-19_17-33-55.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/2520869/OpenBCI-RAW-2018-10-19_17-33-55.txt)
Here are two file examples! Most of the electrodes are railed. 

[image]

Here is the set-up of the electrodes. Is there any other information that would be helpful?

retiutut, 2019-02-22:
@hfuhr Is this issue still happening with GUI 4.1.0?

The playback files you have shared do not work anymore, due to improvements.

closed 2019-05-04 by retiutut

michal-demecki, 2020-12-19:
@hfuhr, how have you solved this issue?

retiutut, 2020-12-19:
@michal-demecki please contact official support or make a post in the forum. This issue is no longer relevant, and is related to hardware, no the GUI.