OpenBCI/OpenBCI_GUI #390: Data Not Streaming openBCI GUI when connected to Cyton through Wifi

HamzaHummam, 2018-10-22:
## Problem
No Streaming when OpenBCI GUI is connected to Cyton Through Wifi
The data is streamed on OpenBCI GUI when connected to cyton through serial port but when connected through wifi the data is not streamed , I have tried all versions of Wifi Firmware and currently loaded the latest firmware 2.0.5 and I also have tried several versions of GUI and HUB.Any help will be appreciated
## Expected 
Live Data Streaming when OpenBCI GUI is connected through wifi to the Cyton Board
## Operating System and Version
Windows 10
## GUI Version
3.3.1 , 3.3.2 and 3.4.0
## Running standalone app
I have tried through the standalone app and through processing both
## Type of OpenBCI Board
Cyton
## Are you using a WiFi Shield?
Yes

ahsanic, 2018-10-24:
I am having the same problem.

It goes to the 2nd screen but its not streaming any data?

Any Idea?

andrewjaykeller, 2018-11-01:
Are you using WiFi direct or WiFi over a router?

andrewjaykeller, 2018-11-01:
I’ve seen this happen if the battery isn’t fully charged

HamzaHummam, 2018-11-23:
We have used GUI version 3.2.1 , Hub Version 1.3.2 and the data is only streaming on Windows 7 Platform we had tested the same version on different platforms such as Windows 10 and Mac OS 10 but the data was not streaming ..... its only working on windows 7 what could be the possible bug can't we solve it

jps2000, 2018-11-23:
Forget all those outdated versions. Always upgrade to latest versions

HamzaHummam, 2018-11-23:
the Latest Version even do not respond as soon as I click on the Start System Button .... I have tested them on Windows 10 and Mac OS 10 but they even do not respond as soon as I click on Start System Button 
Sometimes it stuck in the beginning when I execute the exe file of the GUI .... shows some kind of exception error 

> Forget all those outdated versions. Always upgrade to latest versions

jps2000, 2018-11-23:
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.0.3-beta.0 ??
Dongle is assigned to a com port? Hub started before?
If y then I also do not know
I had the problem that I expected to run latest version but it was not - my fault.

closed 2019-06-10 by retiutut