OpenBCI/OpenBCI_GUI #391: No Data Streaming On OpenBCI GUI when Connected through Wifi

HamzaHummam, 2018-10-22:
## Problem
When I connect openBCI GUI to Cyton Board through Serial Port It works fine but when I connect them through Wifi Shield there is no data streaming I have replaced the battery , I have tried several firmware versions , GUI versions and Hub Version but the problem remains the same  
## Expected 
Live data streaming from cyton board through wifi shield
## Operating System and Version
Windows 10

## GUI Version
I have tried several versions like 3.3.4 , 3.3.1 ,3.3.2

## Running standalone app

I have tried standalone as well as through processing 

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

Yes

closed 2018-10-22 by HamzaHummam

not-Ian, 2019-02-07:
HI Hamza, did you find a solution to your problem? I am also having a problem streaming from a Wifi shield on Windows 10

not-Ian, 2019-02-07:
For those in the future: I found my issue.

I was connected to my private home network. Here are the changes I made, I'm pretty sure only #2 and #3 were necessary:

1. classify my network as a private network under network properties
2. Turn off firewall using windows defender
3. run terminal as an administrator

Then my python script worked!