OpenBCI/OpenBCI_GUI #398: Accelerometer display in playback in 4.0.2

jps2000, 2018-11-07:
## Problem
It seems that accelerometer  displays all (from begin to end of the file). However it is difficult to find corresponding time series.

## Expected 

A marker in acc display moving together with the  time bar would be helpful.
Start  should clear the time series display.  After moving the time bar and pressing start again previous time series is still shown and continues with new. That is confusing.

## Operating System and Version

Windows 7 professional 64bit

## GUI Version

4.0.2

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

jps2000, 2018-11-12:
Note that chip has flipped  pcb side between cyton and ganglion.  ( x and z axis changed sign)

retiutut, 2018-11-20:
> Note that acc chip has changed pcb side between cyton and ganglion. 👍 

The following pictures document this issue. Cyton and Ganglion were placed side by side using GUI on two machines, then rotated in the same way. The GUI shows different results: 

- Flat and facing up
[image]
[image]

- Rotation 1
[image]
[image]

- Rotation 2
[image]
[image]

jps2000, 2018-11-20:
Yeah thats what I discovered.
Assuming Cyton as the  reference, Ganglion should deliver same acc output (preferably).

retiutut, 2018-11-20:
@andrewjaykeller Can I go ahead and fix this using Cyton as a reference?

andrewjaykeller, 2018-11-20:
So X and Z are multiples by -1?

retiutut, 2018-11-20:
> So X and Z are multiples by -1?

Z can be inverted for Ganglion.

X and Y may need to be swapped?

jps2000, 2019-02-11:
Solved in 4.1.0
Thank you

commit f9098250 (2019-04-02): Apply Fix for Ganglion+WiFi Fixes #398
closed 2019-02-11 by jps2000