OpenBCI/OpenBCI_GUI #399: Playback from file in 4.0.2 error

jps2000, 2018-11-08:
## Problem
Version 4.0.2  write one column more in ganglion
Example:
previous version :
%OpenBCI Raw EEG Data
%
%Sample Rate = 200.0 Hz
%First Column = SampleIndex
%Last Column = Timestamp 
%Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
0, 153.43, 646.95, 437.44, -34.03, 0.000, 0.000, 0.000, 03:10:01.012
1, 138.93, 629.45, 457.75, -46.83, 0.000, 0.000, 0.000, 03:10:01.014
2, 134.32, 632.02, 440.07, -43.25, 0.000, 0.000, 0.000, 03:10:01.014

version 4.0.2
%OpenBCI Raw EEG Data
%Number of channels = 4
%Sample Rate = 200.0 Hz
%First Column = SampleIndex
%Last Column = Timestamp 
%Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
0, 14.27, 15.53, 10.35, 15.34, 0.000, 0.000, 0.000, 14:45:42.454, 1541598342454
1, 12.31, 12.26, 11.11, 16.43, 0.000, 0.000, 0.000, 14:45:42.469, 1541598342469
2, 11.52, 13.29, 8.38, 12.38, 0.000, 0.000, 0.000, 14:45:42.469, 1541598342469
3, 14.91, 16.07, 13.62, 13.64, 0.000, 0.000, 0.000, 14:45:42.484, 1541598342484
4, 13.79, 16.78, 17.93, 20.13, 0.000, 0.000, 0.000, 14:45:42.484, 1541598342484

so there is one column more. ( Aux data???)
Despite the correct number of channels reported in the header (4) OpenBCIGUI reports 8 channels with crap accelerometer reading (1541598342484 reported as x).
I  edited the file and by removing last column it works as expected
Accelerometer  data are ok then however  not moving smoothly as time series does. ( rattling)

## Expected 

Playback should work with new recordings as well as with old recordings

## Operating System and Version

W7 professional

## GUI Version

4.0.2

## Running standalone app

downloaded app 

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

andrewjaykeller, 2018-11-08:
Hi! We added the new column as a much more accurate time stamp in ms! Are you saying you can’t use a playback file recorded in 4.0.2 on ganglion?

jps2000, 2018-11-08:
Unfortunately yes.
Files recorded with last release GUI works.  Recordings with 4.0.2 don´t.
Deleting last column in those files let it work again.
%Number of channels  line  has no effect.
I just issued another problem in ganglion code ( see there)

retiutut, 2018-11-14:
> OpenBCIGUI reports 8 channels with crap accelerometer reading (1541598342484 reported as x).

**Woah! That doesn't sound good.**

Looking into this issue also...

andrewjaykeller, 2018-11-14:
Hi @jps2000 this was just fixed for 4.0.3 on mac and windows! https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.0.3-beta.0

jps2000, 2018-11-15:
great
I will  check it out today- thanks

commit 454827a6 (2018-11-14): Merge pull request #404 from andrewjaykeller/master

FIX: mojave issue and ganglion playback closes #399 and #402
commit 8c3942bd (2018-11-14): FIX: mojave issue and ganglion playback closes #399 and #402
closed 2018-11-15 by jps2000