OpenBCI/OpenBCI_GUI #400: Data Logging start / stop

jps2000, 2018-11-09:
## Problem
Data logging start works as expected. Data logging stop does not. one need to stop system that a file is written and closed. This is similar to #2 but not solved by #38 etc. Files of several hours recording are not  easy to handle in playback ( take very long to open etc) 

## Expected 
I would expect to stop a recording and write the file as soon as stop is pressed. This avoids long recordings that can not (hardly) playback.
Ideally there is a pause function as well ( Start continue recording in same file) but that is nice to have

## Operating System and Version
W7 professional

## GUI Version

4.0.2

## Running standalone app

downloaded app

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

retiutut, 2018-11-09:
Looking into this...

retiutut, 2018-11-13:
@jps2000 Thank you for making an issue for this! Let's discuss the features you would like added.

Currently we can Start/Pause using the "Start/Stop Data Stream" button. As you explained, we need a way to completely stop a recording and save the log file without stopping the system.

I propose the following solution:
- [ ]**Add a Stop Recording button to Live mode for Ganglion/Cyton**
- [ ] Stop Recording Button will write file and prompt user to select new log file info, just like the system control panel

jps2000, 2018-11-13:
start stop data stream making start pause is fine. Pressing  stop in the network screen should  stop recording and save the log file. data stream may continue. pressing start again should start with new log file.
so second proposal is fine. No new button is needed.

retiutut, 2018-11-13:
That button is for networking only. This issue involves file operations.

jps2000, 2018-11-13:
I do not understand "networking only"   but I am happy with any improvement that is self explaining

retiutut, 2018-11-13:
Or maybe call it a "Save Log File" button? I made a picture to help clarify what I propose.
[image]

jps2000, 2018-11-13:
what is start stop networking doing?
When I select time series  for example in stream 1 then start networking  starts log file now.  Button turns to red and is called Stop. Pressing Stop however has no effect.
That is what I understand  now.
I suggest that stop  close the log file.
Is this contradiction to something?
Do you need  a start/ stop button for every stream as alternative?

jps2000, 2018-11-13:
New file name can include actual time stamp

retiutut, 2018-11-13:
Start/Stop Network stream has nothing to do with the log file!

jps2000, 2018-11-13:
But what is it doing?
when is the logfile starting now?

retiutut, 2018-11-13:
In the Networking Widget, the Start/Stop Networking button allows you to send data to another program.

Currently, the **log file starts recording to file when you press "Start Data Stream",** and pauses when you press "Stop Data Stream".

jps2000, 2018-11-13:
Will re- check tomorrow (ganglion). That is not according to my experience but I may be wrong....
I used to select time series in stream 1 and pressed start after start data stream.  So you are saying thats useless.

I want to start/ stop datastream ( to verify what is coming) and start/stop logfile on top of. The latter  should work only when data stream is running. It would be even better when stop datastream is only allowed once logfile has stopped.

Anyhow... make something that is understandable for IQ 100 guys.

jps2000, 2018-11-14:
Have tested it and of course you are  right. I interpreted the networking widget as universal I/O tool that also includes  log file writing.

andrewjaykeller, 2018-11-15:
@conorrussomanno what do you think?

retiutut, 2018-12-16:
https://github.com/OpenBCI/OpenBCI_GUI/issues/30

jps2000, 2019-02-23:
From #443 :
For example sleep studies etc need analysis of recordings in the magnitude of hours.
Such long files can not be handled now.
A remedy may be a recording scheduler to limit length of recordings.
(start a new recording (file) after x minutes)

closed 2019-03-15 by jps2000

retiutut, 2019-03-18:
Sorry for the delay on this issue.

> For example sleep studies etc need analysis of recordings in the magnitude of hours.
> Such long files can not be handled now.
> A remedy may be a recording scheduler to limit length of recordings.
> (start a new recording (file) after x minutes)
> 

@jps2000 Are you sure this needs to be closed? 

Or maybe rename it to "Add support for longer recordings in GUI"?

jps2000, 2019-03-19:
I´d better open a new one on that