OpenBCI/OpenBCI_GUI #402: GUI or Hub cannot see Ganglion, if macOS user upgrades to Mohave (10.14)

wjcroft, 2018-11-14:
## Problem

Explain the problem

## Expected 

In a perfect world, what do you expect to happen.

## Operating System and Version

macOS/Windows/Linux

## GUI Version

The version is displayed on startup.

## Running standalone app

Are you running the downloaded app or are you running from Processing 3

## Type of OpenBCI Board

Cyton/Cyton+Daisy/Ganglion

## Are you using a WiFi Shield?

Yes/No

wjcroft, 2018-11-14:
All GUI versions up to 4.0.2 are seeing this. Both standalone and Processing initiated. No Shield. See these threads on forum:

http://openbci.com/forum/index.php?p=/discussion/1876/no-ganglion-bluetooth-connection-after-macos-mohave-update#latest

http://openbci.com/forum/index.php?p=/discussion/1861/ganglion-on-mac-mohave-no-bluetooth-connection#latest

Regards, William

andrewjaykeller, 2018-11-14:
Hi @wjcroft !! So nice to see you one Github! I'm working on a patch right now!

andrewjaykeller, 2018-11-14:
Hi @wjcroft please point users to:

https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.0.3-beta.0

closed 2018-11-14 by andrewjaykeller

mr-yamraj, 2018-11-23:
Hi! @andrewjaykeller it is still showing a blank screen.
[image]